hamuchiwa/AutoRCCar #85: The test and training accuracy keeps on changing with the same collected data.

avisrivastava254084, 2018-03-23:
I have made sure that the training images and the training data are the ones from my recent collection(not the old ones). I got the training accuracy of 89% and test of 60 percent. When I re-ran the mlp_training.py, the accuracies dropped significantly to 60 and 43, respectively. Now, when I ran the code for the third time, the accuracy was again dropped to 76 and 36. What might be the reason for this?

hamuchiwa, 2018-03-30:
When you train the model, parameters (weights) are randomly initialized, and the data is shuffled into train and validation set. You probably need more data, collect more images and give it a try.

anandak95, 2018-04-04:
I have the same problem of varying accuracy. i tried training the model only the forward motion and the training accuracy where as follows. (99,99), (100,100), (77,77), (55,55). i have a total dataset of 3970 selected images just for a forward motion. i believe that is more than enough. Also the car goes left or right sometimes  even though we have trained it only for the forward motion.

Hope you can solve this. Regardless of all of these its still such a great work sir.
Thankyou.

hamuchiwa, 2018-04-05:
@anandak95 Training a model sometimes can be very tedious. If using half of the image doesn't generate a good result, you can try to tilt down the camera angle a little to make sure only track is filled in the image. Also you can try to tune the parameters if that helps.

closed 2018-04-16 by hamuchiwa