hamuchiwa/AutoRCCar #86: Car not on track despite the amazing training results(accuracy)

avisrivastava254084, 2018-03-23:
I have trained the car to 90% training and 60% testing accuracy. However, the car is not able to run successfully run on the track. Here are the videos: 1. https://photos.app.goo.gl/25NyShp9mrt47B0J3
2. https://photos.app.goo.gl/TDsIPjCbnqeLMhPS2

Any insights?

tham1996, 2018-03-28:
Sorry bro I can't answer your question, but may I know how do you manage to let the car move? Mine jus won't move after opening the rc_driver.py, although i only have a test accuracy of 30%, does that related? Thank you in advance!

HowToBeta, 2018-04-07:
same problem here also

geek1111, 2018-04-18:
@avisrivastava254084 Slow down your Car's speed. By adding more gears or using variable resistor cut down access voltage.

closed 2018-06-29 by hamuchiwa