hamuchiwa/AutoRCCar #87: Raspberry Display

hsharma27, 2018-03-23:
I am using VNC Player for displaying raspberry display and it is giving me error for picamera. So where am I going wrong ? should i go with hdmi display or VNC Player ? Thanks in advance

Gooose1909, 2018-03-23:
Raspberry display in the sense for displaying the screen of raspberry pi
If yes
I would recommend u to use remote desktop connection app
Which is inbuilt in windows

hsharma27, 2018-03-23:
But then why is it giving me error in picamera (Using VNC Player and Raspbian Jessie).

Gooose1909, 2018-03-23:
If u are facing error while raspistill
Then this might be the reason :
The preview is a superimposed video layer on top of whatever desktop or
console you are using and is therefore not available for VNC to
send/receive.

You could try network streaming.

closed 2018-06-29 by hamuchiwa