hamuchiwa/AutoRCCar #88: Remote Control of the Car..

Cagriyoruk, 2018-03-24:
I am having a problem with controlling the car via python (keyboard).I found and soldered the wires I have to solder.Then I upload the code to the Arduino.After that, I opened the rc_control_test and tried to run the code.There were no problems with forward and right but the reverse and left didn't work.I tried everything in the threads.The level shifting also voltage dividing.But couldn't solve the problem.I did nearly 15 possible circuit combinations.But no result.I hope that you guys can give me hints.
[image]
[image]

Cagriyoruk, 2018-03-24:
They are not negative.But there is something wrong.When is feed the circuit with the 5v Arduino output, The open circuit voltage is nearly 5v.When I press just forward right has 0.44V.Left and reverse have 0v.When I press left or reverse the 5v is coming but they're not reacting.No LED light no car movement.But when Measuring it I am seeing the 5v.In total just forward, right and forward right is working :/. I tried to feed it with both the Arduino and the 2 AA batteries total 3.1V.When I fed it with the 3.1V they couldn't communicate with the Arduino and no car movement again.
[image]

Gooose1909, 2018-03-24:
Thats the problem with current model remote
No proper way of interfacing arduino with it
I tried 3 different remote and failed
I really recommended you to use an rf transmitter and reciever

Cagriyoruk, 2018-03-24:
There are some people who had the same issue with the remote.But I think they've solved the problem.I tried to do the same thing but again no result.
Here is an example:
https://arduino.stackexchange.com/questions/21205/controlling-an-rc-car-using-arduino
I tried to implement the solution but didn't work out at all.
I don't think I can use the rf transmitter and receiver in this case.My car has some problems I tried to communicate with the Bluetooth module and made a direct connection through Arduino pins and the motor.

Gooose1909, 2018-03-24:
http://blog.davidsingleton.org/nnrccar/
Try this method and say your results

Cagriyoruk, 2018-03-28:
I tried the method but it didn't work out.But I found another solution.I use optocouplers as a switch.And now all possible combinations are working.Here is the solution I found.I used 5k resistors instead of 10k.Built the circuit 4 times for each direction.Got the VCC from Arduino's 5v.If you have any questions feel free to ask.

[image]

[image]

closed 2018-03-28 by Cagriyoruk

Moha-alfridi, 2018-04-16:
Can you please share a clear schematic, I am having this problem, with my model.

signal input and signal GND, is it the remote signal?
what is the AVR?

Instead of using optocouplers , can I build it using mosfit ?

kxrf97, 2019-02-20:
@Cagriyoruk i have the same controller as yours, do you mind to help me with setup