hamuchiwa/AutoRCCar #89: Question about controlling the RC car

tkajbaje, 2018-03-27:
I am trying to control the car using my laptop but when I run rc_control_test.py I get the following error

> No protocol specified
No protocol specified
No protocol specified
xcb_connection_has_error() returned true
Home directory not accessible: Permission denied
No protocol specified
xcb_connection_has_error() returned true
Home directory not accessible: Permission denied
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave

hamuchiwa, 2018-03-30:
Is it "permission" related?

tkajbaje, 2018-03-30:
`xhost si:localuser:root
`
This line of code solved my problem

aakashm301, 2018-03-31:
@tkajbaje what you did for connecting the rc with laptop. I also cannot run rc_control_test.py 
**Having following error**
C:\Python27\python.exe C:/Users/Aakash/Desktop/AutoRCCar-master/test/rc_control_test.py
Traceback (most recent call last):
  File "C:/Users/Aakash/Desktop/AutoRCCar-master/test/rc_control_test.py", line 69, in <module>
    RCTest()
  File "C:/Users/Aakash/Desktop/AutoRCCar-master/test/rc_control_test.py", line 12, in __init__
    self.ser = serial.Serial('/dev/tty.usbmodem1421', 115200, timeout=1)
  File "C:\Users\Aakash\AppData\Roaming\Python\Python27\site-packages\serial\serialwin32.py", line 31, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Users\Aakash\AppData\Roaming\Python\Python27\site-packages\serial\serialutil.py", line 240, in __init__
    self.open()
  File "C:\Users\Aakash\AppData\Roaming\Python\Python27\site-packages\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port '/dev/tty.usbmodem1421': WindowsError(3, 'The system cannot find the path specified.')

Process finished with exit code 1

Gooose1909, 2018-03-31:
/dev/.... File location is invalid
Open arduino software and connect your arduino and check for the port eg
:COM1 . It will be on the bottom right of your arduino software. Replace
/dev... To that port value. It will fix it

tkajbaje, 2018-03-31:
@aakashm301 Gooose1909 is right,replace '/dev/tty.usbmodem1421' with the port you are using to connect arduino to your pc e.g. COM1.

tkajbaje, 2018-03-31:
Anyone has modified rc_driver.py code for opencv3.0 ?

tkajbaje, 2018-03-31:
Hello Everyone,so I modified rc_driver.py for opencv3.0 as guided by @hamuchiwa in another thread.but when I run rc_driver.py and connect with the car,my car moves out of track,it seems that the rate at which the frames are received from the camera and the speed of the car does not match,is there any way I can stop the car for some time unless a new frame is received and prediction is made on that so as to synchronise the speed of car and speed of receiving images from picam

aakashm301, 2018-03-31:
Thank you so much guys for your help, it worked.

tkajbaje, 2018-03-31:
So I solved the synchronization problem by adding stop() after every move of RC Car in rc_driver.py.and changing the value of delay in Arduino sketch to 150 ms .the car now moves slowly (actually very slowly) but runs on track.

aakashm301, 2018-04-01:
Can you please share the code because I will also be facing the same problem because my RC runs really fast.

tkajbaje, 2018-04-01:
Here is the link to the modified code. 
https://drive.google.com/drive/folders/10iQPYrRAZ98uMOC_vT-8OojYjL9NWkYj?usp=sharing

aakashm301, 2018-04-01:
Thank you so Much. One last question what you did for running rc_control_test.py after changing the specified address of port in Arduino it worked but nothing happened. It just ran and keeps running no window popped up nothing.

Gooose1909, 2018-04-01:
No pygame window opened??

aakashm301, 2018-04-01:
No nothing happened.

Gooose1909, 2018-04-01:
pygame.init()
    size = [70, 50]
    screen = pygame.display.set_mode(size)

 Makw sure this line is there in your code

tkajbaje, 2018-04-01:
I have added the rc_control_test.py script to above the above link (https://drive.google.com/drive/folders/10iQPYrRAZ98uMOC_vT-8OojYjL9NWkYj?usp=sharing) change the port no. to your desired port and run the script.

aakashm301, 2018-04-02:
Thank you so much it worked

aakashm301, 2018-04-03:
Okay so now there is error in connecting the. I am successfully able to control car with keys but i cannot connect my raspberry pi with laptop.

When I try connecting with this IP and port (192.168.0.117', 8000) it gives me this error

C:\Python27\python.exe "C:/Users/Aakash/Desktop/AutoRCCar-master/AutoRC Car/collect_training_data.py"
Traceback (most recent call last):
  File "C:/Users/Aakash/Desktop/AutoRCCar-master/AutoRC Car/collect_training_data.py", line 173, in <module>
    CollectTrainingData()
  File "C:/Users/Aakash/Desktop/AutoRCCar-master/AutoRC Car/collect_training_data.py", line 18, in __init__
    self.server_socket.bind(('192.168.0.117', 8000))
  File "C:\Python27\lib\socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 10049] The requested address is not valid in its context

Process finished with exit code 1

SO i tried another IP and PORT ('127.0.0.1', 5560)
the server starts but it gives error CONNECTION REFUSED

Gooose1909, 2018-04-03:
Replaces that IP with your laptop's Ip address
U can find it running "ipconfig" command on CMD

aakashm301, 2018-04-04:
Thank you that worked but I am getting another error.
I streamed the data and connected the arduino with laptop as it says that we have to control the car with keys while collecting data. But when I start streaming it said **Collecting data** and then
C:\Python27\python.exe C:/Users/Aakash/Desktop/AutoRCCar-master/computer/collect_training_data.py
Start collecting images...
Traceback (most recent call last):
  File "C:/Users/Aakash/Desktop/AutoRCCar-master/computer/collect_training_data.py", line 172, in <module>
    CollectTrainingData()
  File "C:/Users/Aakash/Desktop/AutoRCCar-master/computer/collect_training_data.py", line 35, in __init__
    self.collect_image()
  File "C:/Users/Aakash/Desktop/AutoRCCar-master/computer/collect_training_data.py", line 59, in collect_image
    image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.CV_LOAD_IMAGE_GRAYSCALE)
AttributeError: 'module' object has no attribute 'CV_LOAD_IMAGE_GRAYSCALE'

Process finished with exit code 1

tkajbaje, 2018-04-04:
Replace 'CV_LOAD_IMAGE_GRAYSCALE' with "IMREAD_GRAYSCALE" without the quotation marks.

closed 2018-04-23 by tkajbaje