hamuchiwa/AutoRCCar #90: Serial port about could not open port 'COM4(Arduino/Genuino Uno

Maxxiezhu, 2018-04-01:
I'm sure that I'm not mistaken the serial port of my PC.
But,PC show this when I  run the "rc_control_test"
Please tell me why, thank you very much.
[image]
[image]
[image]

Gooose1909, 2018-04-01:
Running CMD as admin might solve the problem

hamuchiwa, 2018-04-01:
@Maxxiezhu 括号的内容去掉，只留‘COM4’试一下

Maxxiezhu, 2018-04-01:
@hamuchiwa 我明白了，之前是我错了，应该是COM4，但是还是提示无法打开通讯。还有个问题是用python_IDE环境下打开那个rc_control_test文件吗？

Gooose1909, 2018-04-01:
@maxxiezhu
Changing ur port number may fix it

Maxxiezhu, 2018-04-01:
@Gooose1909 Unable to modify...

Gooose1909, 2018-04-01:
From the Windows desktop, right-click on My Computer. Then Properties >
Hardware, Device Manager > Ports (COM & LPT). There will be an entry like
"USB Serial Port (COM4)". This means serial comm port 4 is the one in use.

Change it to port 8 or something thats not in use then try it

hamuchiwa, 2018-04-01:
@Maxxiezhu 我用的是pycharm，不过终端打开应该也行。我觉得可以先试一下Arduino自己带的样例文件，看看文件能不能刷到板子上。Gooose1909说的那些方法试过了没，管不管用？

Gooose1909, 2018-04-01:
My Computer -> Manage -> Device Manager -> Ports -> Communications Port -> Port Settings -> Advanced -> COM Port Number
Sorry this is the right directory to change port value

Maxxiezhu, 2018-04-01:
ok,,I  change it to port8. And the PC Show this. And the Arduino's signal led flashed.
[image]

The desktop create a file (__pycache__)
[image]
[image]

What do I do next?How do I know it  communicate successfully?
@Gooose1909 @hamuchiwa

Maxxiezhu, 2018-04-01:
@hamuchiwa 我Arduino板子是正常的可以下载程序进去的

aakashm301, 2018-04-01:
I was also facing the same problem on the laptop I changed the address mine is COM6. And I ran the code in pycharm but nothing happened it just showed that it is running.

hamuchiwa, 2018-04-01:
@Maxxiezhu 先禁用COM4，然后再重新启用~?

Maxxiezhu, 2018-04-02:
@hamuchiwa @Gooose1909 
And I've already installed pyserial.
[image]

But, I had this problem when I was running the code that "rc_control_test". 
[image]

How to deal with this situation?

Maxxiezhu, 2018-04-02:
@hamuchiwa 你好，上面问你的那些问题处理之后没有再次出现了。但是我想问一下。
1. 运行那个rc_control_test测试文件，怎样才说明测试正常？成功测试？
2.我运行完之后命令窗口弹出来，是这样的
[image]
命令窗口里面没有任何信息，也不能在命令窗口输入任何东西这样正常吗？不过我测试文件运行成功之后，同时Arduino板通信口tx闪烁了一下
3.还有就是为什么运行文件之后，桌面生成一个文件夹？里面是一个pyc文件
[image]
麻烦你给我解答一下这些问题，十分感谢。

Maxxiezhu, 2018-04-02:
@hamuchiwa 方向键是RC遥控器上面的按键对吧？

hamuchiwa, 2018-04-02:
@Maxxiezhu 键盘上的方向键，等于是拿键盘当遥控器使。

Maxxiezhu, 2018-04-03:
@hamuchiwa 
你好，我运行之后，出现这个命令窗口，并且我也把光标放在窗口外，然后用计算机键盘分别按了上下左右，小车没有任何反应。这个是什么问题？四个按键按下去PC也没有任何显示。还是说我按按键的方法错了？
[image]

Maxxiezhu, 2018-04-03:
@hamuchiwa 
你好，加了那两行代码之后，成功连接上Arduino了。但是有个问题。我运行代码之后。弹出来了命令窗口和一个pygame的窗口。
[image]

然后我在pygame出口处，按下我键盘上面的上键下键左键右键都不行。他这样提示我的。
[image]

不知道怎么办。麻烦你帮我一下

Maxxiezhu, 2018-04-03:
@hamuchiwa 
你好之前问你的问题，我尝试了一下，我去掉了chr直接改成读取数字之后，上面的问题没有出现了。然后我再点击键盘上下左右，看到命令窗口显示了按键的方向。
[image]
[image]

但是有问题出现了
1.Arduino对应的引脚没有变成低电平去触发小车运动。但是当我每按下一个键盘按键时Arduino的RX串口都闪烁了一下，说明Arduino已经读取了Pc发送的信息，但是小车就是不能运动。是因为我去掉了char的原因吗？

Maxxiezhu, 2018-04-04:
你好，我找到解决方法了。我已经解决了。谢谢你后面加上这个函数就没有问题了。测试成功！

hamuchiwa, 2018-04-04:
@Maxxiezhu 方便的话能不能留一下系统和环境的配置？以后要有其他人遇到类似问题我就把他们指到这儿。再想问一下encode()函数的默认参数是什么，会不会是因为中英文系统而导致的不同？

closed 2018-06-07 by hamuchiwa