hamuchiwa/AutoRCCar #92: Car running without input from arduino

gamer2199, 2018-04-03:
The script rc_control_test.py is running but i am not sure whether its taking my input or not. The remote buttons have 4 pins connected to the board of which i have soldered wires to two pins because connecting the wire to single pin doesn't make the car work. After connecting to the 2 pins the car goes only in forward direction. Urgently help required!! 
[image]
[image]

Gooose1909, 2018-04-03:
Let me guess you are from india??
I had to deal with the same type of cicuit to interface with arduino
I failed cos the other 2 pins runs on -ve voltage which ur arduino cant
supply
Thats what your problem is

gamer2199, 2018-04-03:
Any solution for this?

Gooose1909, 2018-04-03:
I made myself a new rc car using L298n motor driver with RF transmitter and
reciever
Works like a charm

gamer2199, 2018-04-03:
i guess your transmitter is the remote and you have connected it to the arduino?

Gooose1909, 2018-04-03:
Yeah
Arduino + transmitter
And reciever + arduino to control motor driver for car movement

gamer2199, 2018-04-03:
wait you have 2 arduinos in the project?

Gooose1909, 2018-04-03:
Yup

gamer2199, 2018-04-03:
Can you post an image if possible? it would give me a clear understanding

Gooose1909, 2018-04-03:
Give me a hour will post the pic soon

gamer2199, 2018-04-03:
Thanks a lot..it will really help me :)

gamer2199, 2018-04-03:
Hey....is the project ready? i am in a hurry to see it because my submission are approaching and i am stuck

gamer2199, 2018-04-05:
sorry to bother you but can you help me with the code?

Gooose1909, 2018-04-05:
Yeah

gamer2199, 2018-04-05:
are there a lot of changes?

Gooose1909, 2018-04-05:
Nope
His code remains the same
I just made my own RC car
Thats all

gamer2199, 2018-04-05:
thats awesome. Did you follow any guide online on making the RC car?

Gooose1909, 2018-04-05:
Nope
Basic arduino programming is fine

gamer2199, 2018-04-05:
Can you post your arduino code or is it the same as the one provided with this project?

Gooose1909, 2018-04-05:
Ok

gamer2199, 2018-04-05:
Thanks in advance!

tkajbaje, 2018-04-05:
I have same type of remote for my rc car what I did was I used a 4066 IC for controlling it using arduino it is a quad bilateral switch IC.You have to connect the buttons using wires to every input and output of the IC 

[image]

for eg connect two wires of forward button with input1 and output1 of ic and connect Control A to arduino that pin will control the forward switch of the car.you just have to change the arduino sketch a little bit to make the desired pin HIGH for a duration of time.the IC consists of four switches which are in open condition initially when the control pin of the corresponding switch is provided with 5v supply it goes into closed state.This seemed to be a simple solution to me.

gamer2199, 2018-04-05:
have you tried this solution? for now my group is deciding on doing what goose said but i am still gonna give this a try

tkajbaje, 2018-04-05:
yes I have tried this solution,I can control my car through pc using this method.

gamer2199, 2018-04-05:
do you have a picture of the connections right now?

tkajbaje, 2018-04-05:
[image]
[image]
[image]
[image]
[image]
[image]

gamer2199, 2018-04-05:
Thanks a lot..one last question...the power to the remote control board is through the battery right?

tkajbaje, 2018-04-05:
Yes.You have to connect only the control and power pins of the IC to Arduino.The remote is powered by separate cells.

gamer2199, 2018-04-05:
You are a life saver!!! and Can you please post your arduino code? Sorry for asking so much but it will help me a lot

tkajbaje, 2018-04-05:
[archive.zip](https://github.com/hamuchiwa/AutoRCCar/files/1880507/archive.zip)

gamer2199, 2018-04-05:
Thanks a ton!! I hope this works in my project

gamer2199, 2018-04-07:
i tried your solution but its still not working....did you supply negative voltage?

tkajbaje, 2018-04-07:
I did not understand what you mean by supplying negative voltage,negative voltage to the remote control ? Yes,other than soldering two wires to each terminal of the switch I did not change anything in there and you have to power the RC by putting batteries in battery compartment and then connect the wires soldered to switch to IC.The IC needs 5v supply and GND connection.

gamer2199, 2018-04-07:
i am not able to control the car using keypress..i have connected the ic as you said but no avail

tkajbaje, 2018-04-07:
Can you post images of connections ??

gamer2199, 2018-04-07:
[image]
[image]
[image]

gamer2199, 2018-04-07:
no the pygame window doesnt popup

tkajbaje, 2018-04-07:
go to the link I posted above.those files are modified for working with OpenCV 3.0 as guided by @hamuchiwa  in another thread.rc_driver.py is also modified.As you have moved on to testing the car with rc_driver.py I assume that the car now can be controlled using your pc.

gamer2199, 2018-04-07:
yup i can control the car using my pc. I will run the rc_driver.py from your link later and let you know if any errors occur.

gamer2199, 2018-04-09:
self.model = cv2.ml.ANN_MLP_load(file_path)
AttributeError: 'module' object has no attribute 'ANN_MLP_load'

this is the new error thats occuring...my open cv version is 3.1.0

tkajbaje, 2018-04-09:
install OpenCV 3.4.1 for this to work.I followed [this](http://cyaninfinite.com/tutorials/installing-opencv-in-ubuntu-for-python-3/) tutorial.make sure you uninstall all previous versions before installing this one.

gamer2199, 2018-04-10:
I installed open cv 3.4.1 and now its working but the car keeps turning left.Test accuracy was around 72%. Does it have to be higher and what was your accuracy when you ran the script?

tkajbaje, 2018-04-10:
My accuracy was around 80% but I had another issue while I ran the script.The rate at which the frames were streamed from rpi to computer and the speed of the car were not matching so my car also used to keep turning left.I managed to tackle the issue by adding "self.stop()" after every move of the car so as to synchronize.but now the car runs slow(actually very slow).but it runs on track.

gamer2199, 2018-04-10:
have you tried increasing the frame rate?

tkajbaje, 2018-04-11:
Yes,I tried increasing the frame rate it streamed smoothly only in the test(stream_server_test.py) but in rc_driver.py it is still slow.I think the time taken by ml model to predict the output adds to the lag.

gamer2199, 2018-04-13:
i want to increase the speed a bit...its too slow right now. What we I somehow control the speed of motor and increase the frames?

tkajbaje, 2018-04-14:
If you increase the value of variable 'time' in Arduino sketch speed of the car will increase.framerate can be increased in stream client code for rpi.

gamer2199, 2018-04-14:
where do we make the changes in stream client...i am having trouble finding it

tkajbaje, 2018-04-15:
in raspberry pi folder in file stream_client.py change 'camera.framerate' to your desired framerate value.

gamer2199, 2018-04-16:
i changed the frame rate at first but its doesn't do any effect while streaming

closed 2018-04-23 by gamer2199

kxrf97, 2019-02-21:
@gamer2199 tkajbaje solution could solve this issue?

gamer2199, 2019-02-21:
@kxrf97 can you specify which issue exactly? If you are talking about the title, yes it did!

kxrf97, 2019-02-21:
i have a controller almost same as yours

kxrf97, 2019-02-21:
there are 4 pins for each push button

kxrf97, 2019-02-21:
i connect one pin for one button, after i plug into my arduino, only forward and left are functioning. Reverse and right are not receiving any signal at all. May i know which 2 pins you soldier with your wire?

kxrf97, 2019-02-21:
@gamer2199 
[image]
[image]

gamer2199, 2019-02-26:
@kxrf97 sorry for the late reply. I did exactly the same thing that you have done here. Maybe its a connection/soldering issue or input issue? Is the PyGame window opening up?

harshithdayananda, 2019-06-11:
@gamer2199 @tkajbaje need your help. my car was working before with optocouplers but now it stopped working and takes continuous input i.e without input keeps going forward. what do I do?