hamuchiwa/AutoRCCar #93: Car is not recognising stop sign

avisrivastava254084, 2018-04-04:
How should i make the car identify the stop sign?

hamuchiwa, 2018-04-04:
The classifier was trained on real stop sign images, it should work in most of the cases. However the traffic light classifier only works for the one used in the video.

avisrivastava254084, 2018-04-04:
> The classifier was trained on real stop sign images, it should work in most of the case

I am sorry, I did not get that.
Also, I printed out the stop sign(the standard one) and pasted it on an ice cream stick. Is this right?
How else would you do? Could you please paste the image and a probable size that should be used?

hamuchiwa, 2018-04-04:
1. regular stop sign image should work
2. can you remind me where you saw "point pin"
3. it's correct either way (including or not including stop sign)
4. the stop sign classifier works out of box, if you plan to train your own, I found that tutorial very helpful

avisrivastava254084, 2018-04-05:
@hamuchiwa 
Sorry, I meant point pi towards the screen and not point pin.

So, if I want to use your code to train my car stop at the Stop sign, what is the procedure?

hamuchiwa, 2018-04-05:
@avisrivastava254084 Ah... I meant put your screen which contains stop sign search result in front of the pi camera. This project doesn't contain haar-cascade classifier training, you need to refer to opencv doc and that tutorial. 
Of course, this is just one option, neural network can be used as well to classify the stop sign and then you can do a sliding window search to find it.

avisrivastava254084, 2018-04-06:
@hamuchiwa, I see you've an XML for stop sign and that you use it in rc_driver.py
I have read the tutorial that you have mentioned in step #5. Now, as you said, I need to point picam to stop sign, which file should I run to make my code understand all of this positive and negative samples?
Basically, as a rookie, I am asking you the steps to incorporate stop sign feature. I have understood the flow of control because of the documentation but I am not able to comprehend how to apply it in this project. A simple steps guide would be helpful. 

Thanks.

hamuchiwa, 2018-04-06:
For using a classifier, you can take a look this [OpenCV tutorial for face detection ](https://docs.opencv.org/3.4.1/d7/d8b/tutorial_py_face_detection.html), replace the xml in the tutorial with the stop sign xml.
For training, it's not included in this project and there is no .py file for this. You have to collect positive (stop sign) and negative (not contain stop sign) images and that tutorial and OpenCV doc can guide you through this.

closed 2018-04-16 by hamuchiwa