hamuchiwa/AutoRCCar #95: ValueError: total size of new array must be unchanged when image is changed

tham1996, 2018-04-07:
I am trying to make a standalone car that only use python and RPi based on your code yet this error jump out. After few hours of research I still can't figure out why, hence you're my last hope. Hope you can help me figure out the issue. Thank you in advance.

Error code:

> Traceback (most recent call last):
>   File "/home/pi/Desktop/modified train data collect.py", line 180, in <module>
>     CollectTrainingData()
>   File "/home/pi/Desktop/modified train data collect.py", line 71, in __init__
>     self.collect_image()
>   File "/home/pi/Desktop/modified train data collect.py", line 108, in collect_image
>     temp_array = roi.reshape(1, 38400).astype(np.float32)
> ValueError: total size of new array must be unchanged

Code(I've deleted motor GPIO pin codes to make it easier to look at):
```
import numpy as np
import cv2
import RPi.GPIO as GPIO
import pygame
from pygame.locals import *
import time
import os
import struct
import io
import picamera
import picamera.array
from picamera.array import PiRGBArray

class CollectTrainingData(object):

    def __init__(self):
        self.send_inst = True

        # create labels
        self.k = np.zeros((4, 4), 'float')
        for i in range(4):
            self.k[i, i] = 1
        self.temp_label = np.zeros((1, 4), 'float')

        pygame.init()
        self.collect_image()

    def collect_image(self):

        saved_frames = 0
        total_frames = 0

        # collect images for training
        print 'Start collecting images...'
        image_array = np.zeros((1, 38400))
        label_array = np.zeros((1, 4), 'float')

        try:
            frames = 1
            while self.send_inst:
                cam = picamera.PiCamera()
                cam.resolution = (320, 240)
                cam.color_effects = (128, 128)
                cam.framrate = 24
                time.sleep(2)
                stream = picamera.array.PiRGBArray(cam)
                cam.capture(stream, format='bgr', use_video_port=True)
                # crop the image and grayscale

                frame, ret = np.split(stream.array, [240], axis=0)
                    # select lower half of the image
                roi = frame[120:240, :]

                    # save streamed images
                cv2.imwrite('training_images/frame{:>05}.jpg'.format(frames), frame)

                    # cv2.imshow('roi_image', roi)
                cv2.imshow('image', frame)

                    # reshape the roi image into one row array
                temp_array = roi.reshape(1, 38400).astype(np.float32)

                frames += 1
                total_frames += 1

                    # get input from human driver
                for event in pygame.event.get():
                    if event.type == KEYDOWN:
                        key_input = pygame.key.get_pressed()

                        if key_input[pygame.K_UP]:
                            print("Forward")
                            saved_frames += 1
                            image_array = np.vstack((image_array, temp_array))
                            label_array = np.vstack((label_array, self.k[2]))

                        elif key_input[pygame.K_DOWN]:
                            print("Stop")
                            saved_frames += 1
                            image_array = np.vstack((image_array, temp_array))
                            label_array = np.vstack((label_array, self.k[3]))

                        elif key_input[pygame.K_RIGHT]:
                            print("Right")
                            image_array = np.vstack((image_array, temp_array))
                            label_array = np.vstack((label_array, self.k[1]))
                            saved_frames += 1

                        elif key_input[pygame.K_LEFT]:
                            print("Left")
                            image_array = np.vstack((image_array, temp_array))
                            label_array = np.vstack((label_array, self.k[0]))
                            saved_frames += 1

                        elif key_input[pygame.K_x] or key_input[pygame.K_q]:
                            print 'exit'
                            self.send_inst = False

                            break

            # save training images and labels
            train = image_array[1:, :]
            train_labels = label_array[1:, :]

            # save training data as a numpy file
            file_name = str(int(time.time()))
            directory = "trained_data"
            if not os.path.exists(directory):
                os.makedirs(directory)
            try:
                np.savez(directory + '/' + file_name + '.npz', train=train, train_labels=train_labels)
            except IOError as e:
                print(e)

            print(train.shape)
            print(train_labels.shape)
            print 'Total frames:', total_frames
            print 'Saved frames:', saved_frames
            print 'Dropped frames', total_frames - saved_frames

        finally:
            print("Work laaa")

if __name__ == '__main__':
    CollectTrainingData()
```

hamuchiwa, 2018-04-09:
Hi, can you post the array size of `frame` and `roi`?

tham1996, 2018-04-09:
Sorry if it look stupid, but isn't the array size calculated from

> frame, ret = np.split(stream.array, [240], axis=0)
>                 roi = frame[120:240, :]

I don't really understand about the array thing.

hamuchiwa, 2018-04-09:
Can you add `print (stream.array.shape, frame.shape, roi.shape)` after that second line, I just would like to see what the actual numbers are.

hamuchiwa, 2018-04-09:
Okay, the `roi` is still in color, the shape needs to be `(120, 320)` or `(240, 320)` if using entire image. Make sure you convert `frame` to grayscale before slicing.

tham1996, 2018-04-11:
It worked, thank you so much!

closed 2018-04-16 by hamuchiwa