hamuchiwa/AutoRCCar #98: Controlling the Car

Moha-alfridi, 2018-04-11:
Hi @hamuchiwa ..

I have problem with controlling the car, it only go FWD-LFT and REV-LFT. I am trying to make it go straight at least, I am new to python but I can see the case's are included in the code. can you help please, Thanks

Moha-alfridi, 2018-04-13:
No my connection are good, however, when I run the rc_control_test.py it's only run front-left or back-left. The same happen when I run collect_training_data.py
:This is happening while I am controlling the car from my keyboard:

I have collected data from a left curve track, but I haven't build my mlp training files because I had some errors.

So my question is, should I take fully control of the car while I am collecting the data, or a left curve track will be enough ?

Moha-alfridi, 2018-04-14:
So, I run collect_training_data.py using a left curve track, and obtained the data, however, when I run mlp_training.py I had errors with python 2.7

`('OpenCV:', '3.4.0')
('Numpy : ', '1.14.2')
('Python:', '2.7.10 (default, Oct  6 2017, 22:29:07) \n[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.31)]')
Traceback (most recent call last):
  File "/Users/moha/Desktop/AutoRCCar/computer/mlp_training.py", line 20, in <module>
    X = np.vstack((X, train))
  File "/Users/moha/.virtualenvs/stream_server_test.py/lib/python2.7/site-packages/numpy/core/shape_base.py", line 234, in vstack
    return _nx.concatenate([atleast_2d(_m) for _m in tup], 0)
ValueError: all the input array dimensions except for the concatenation axis must match exactly`

and python 3.6

`OpenCV: 3.4.0
Numpy :  1.14.0
Python: 3.6.5 (default, Mar 30 2018, 06:41:53) 
[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.39.2)]
Traceback (most recent call last):
  File "/Users/moha/Desktop/AutoRCCar/computer/mlp_training.py", line 20, in <module>
    X = np.vstack((X, train))
  File "/Users/moha/.virtualenvs/project1/lib/python3.6/site-packages/numpy/core/shape_base.py", line 234, in vstack
    return _nx.concatenate([atleast_2d(_m) for _m in tup], 0)
ValueError: all the input array dimensions except for the concatenation axis must match exactly`

I used your code in this issue [https://github.com/hamuchiwa/AutoRCCar/issues/60](url) and this error has generated.

Moha-alfridi, 2018-04-14:
now I have the mlp_training.py file run without error, however, the python 3.6 break as the picture shows 

[image]

```
OpenCV: 3.4.0
Numpy :  1.14.0
Python: 3.6.5 (default, Mar 30 2018, 06:41:53) 
[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.39.2)]
Image array shape:  (0, 28800)
Label array shape:  (0, 4)

Process finished with exit code 136 (interrupted by signal 8: SIGFPE)
```

and when I run it with python 2.7 it run without error however, the percentage don't show up

```
('OpenCV:', '3.4.0')
('Numpy : ', '1.14.2')
('Python:', '2.7.10 (default, Oct  6 2017, 22:29:07) \n[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.31)]')
('Image array shape: ', (0, 28800))
('Label array shape: ', (0, 4))

Process finished with exit code 136 (interrupted by signal 8: SIGFPE)
```

hamuchiwa, 2018-04-16:
Your array size is 0, data is not loaded.

Moha-alfridi, 2018-04-16:
I thought the array will start counting from 0, Sorry for wasting your time and thank you so much.

For future reference:
```
dim = 28800
X = np.empty((0, dim))
y = np.empty((0, 4))
```

you should replace 0 with the number of picture you have collected.

closed 2018-04-16 by hamuchiwa

kxrf97, 2019-02-21:
@Moha-alfridi so your car can only go left and forward? could not turn right?