hamuchiwa/AutoRCCar #100: Connection Doubts

kaskagorry, 2018-04-14:
I am following all the steps to make the rc car work, but I have a couple of doubts.

1) I have to connect the arduino to the raspberry, but when I run the computer script, it does not recognize the port '/ dev / ttyACM0'. But, if I connect the arduino to the cable computer, and I run the script on the computer (changing the port to COM3) and the script on the raspberry, I get it to work (my scheduler is w10)
My question is how can I run the program with the arduino connected to the raspsberry pi?

2) I have an ultrasonic sensor HC-SR04, connected to the arduino. I do not have a breadboard to connect it to the raspberry, so can I use the sensor connected to the arduino instead of the raspberry?

Thank you

Moha-alfridi, 2018-04-15:
Hi @kaskagorry,

For your 1st question, if you want to get the same result of @hamuchiwa project, you should connect the arduino "after you upload the code to it" to the computer and control the car from the computer's keys. If you want to connect the arduino to the raspberry pi you should do your own new configuration, it can be couple more steps to get it working.

for your 2nd question, you should connect the HC-SR04 to the raspberry pi and run the `ultrasonic_client.py` on the raspberry pi after you modify it with your computer IP and port.

I hope that help you to make it clear. Best wishes.

kaskagorry, 2018-04-16:
Hi @Moha-alfridi 
If I have not understood you wrong, the arduino has to connect to the computer to be able to handle it.
The only thing that I do not understand, is that in the video of @hamuchiwa it is not appreciated that the arduino is connected by cable to the computer.
I'm not very clear about that step

Thanks

kaskagorry, 2018-04-16:
Ok, but if the arduino is connected by cable to the pc, the robot can not walk much.

In the video it is seen that the robot walks independently without being connected to any computer via usb

closed 2018-06-29 by hamuchiwa