hamuchiwa/AutoRCCar #101: connection issue

omnitrix123, 2018-04-15:
I am confused about the ip's and port in server_bind() and client_connect() functions.
How to get the ip's and ports for both the codes  i.e stream_server_test.py  and stream_client.py?
Also i have an doubt for the car drive..is Rpi's serial interface conected to arduino's serial interface(Rx and Tx)?
Please help.Thank you.

omnitrix123, 2018-04-16:
Thanks a lot @Moha-alfridi .
what about the ip in stream_server_test.py? what ip should i use in bind() ?

Gooose1909, 2018-04-16:
Your laptop / pc IP address

Moha-alfridi, 2018-04-17:
I had this error before and @hamuchiwa solve it by this is a quote 

> I think this is because Python3 handles str and bytes differently, change line 50 stream_bytes = ' ' to stream_bytes = b' ', hope this will fix the issue.

the line number 50 is located in `stream_server_test.py`

Also, follow this quote

> Okay, a few more changes, Python3 treats '\xff\xd8' as string, so change it to b'\xff\xd8' instead. In Python2, bytes and str are interchangeable.

nevertheless, if you are using OpenCV3 don't forget to make these changes in `stream_server_test.py`

> So for future reference for this error
> AttributeError: module 'cv2.cv2' has no attribute 'CV_LOAD_IMAGE_GRAYSCALE'
> you should go ahead to line 33 where it shows
> image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.CV_LOAD_IMAGE_GRAYSCALE)
> and change the CV_LOAD_IMAGE_GRAYSCALE with cv2.IMREAD_COLOR for color stream
> the whole line should be like
> image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.cv2.IMREAD_COLOR)
> 
> and this will show the video stream in gray by using cv2.IMREAD_GRAYSCALE
> the whole line should be like
> image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.cv2.IMREAD_GRAYSCALE)

All of these error were discussed in issue #94 
Best wishes @omnitrix123

omnitrix123, 2018-04-17:
i tried to go to your mentioned url , posted in issue 94 

So, I was trying to figure out the error that I had, and came across a solution here https://stackoverflow.com/questions/19013961/cv2-imread-flags-not-found

but it seems it is not available 
plz help

omnitrix123, 2018-04-17:
how to drive the car during training and how will car run after training?

omnitrix123, 2018-04-17:
help me out with the connections

omnitrix123, 2018-04-17:
@hamuchiwa in your video on youtube what are 2 wireless connections connected on the usb port of rpi please explain

hamuchiwa, 2018-04-18:
@omnitrix123 One is wifi adapter, the other one is for wireless keyboard.

omnitrix123, 2018-04-20:
ty@hamuchiwa

closed 2018-04-20 by omnitrix123