hamuchiwa/AutoRCCar #102: Low FPS while Recording Frames and Key strokes

CT83, 2018-04-16:
I currently using the code below to stream and receive frames from my Raspberry Pi to my PC.

**Client**
```

import io
import socket
import struct
import time

import picamera

client_socket = socket.socket()
client_socket.connect(('my_server', 8000))

connection = client_socket.makefile('wb')
try:
    with picamera.PiCamera() as camera:
        camera.resolution = (640, 480)
        camera.start_preview()
        time.sleep(2)

        start = time.time()
        stream = io.BytesIO()
        for _ in camera.capture_continuous(stream, 'jpeg'):
            connection.write(struct.pack('<L', stream.tell()))
            connection.flush()
            stream.seek(0)
            connection.write(stream.read())
            stream.seek(0)
            stream.truncate()
    connection.write(struct.pack('<L', 0))
finally:
    connection.close()
    client_socket.close()

```
**Server**

```
import io
import socket
import struct

import numpy as np
from cv2 import cv2

server_socket = socket.socket()
server_socket.bind(('0.0.0.0', 8000))
server_socket.listen(0)

connection = server_socket.accept()[0].makefile('rb')
try:
    while True:
        image_len = struct.unpack('<L', connection.read(struct.calcsize('<L')))[0]
        if not image_len:
            break
        image_stream = io.BytesIO()
        image_stream.write(connection.read(image_len))
        image_stream.seek(0)
        file_bytes = np.asarray(bytearray(image_stream.read()), dtype=np.uint8)
        img = cv2.imdecode(file_bytes, cv2.IMREAD_COLOR)

        cv2.imshow('Original', img)
        cv2.waitKey()
finally:
    connection.close()
    server_socket.close()

```
The code is similar to yours, I got it off [PiCamera Repo](https://picamera.readthedocs.io/en/release-1.10/recipes1.html).
But my Frames per second are really slow, almost 3-5 FPS. How can I solve this problem? Decreasing camera resolution didn't help either.
How did you overcome this problem? What were your FPS like? Should I just ignore this can collect more training data to compensate for it?

hamuchiwa, 2018-04-16:
I referenced [PiCamera doc](https://picamera.readthedocs.io/en/release-1.10/recipes2.html#rapid-capture-and-streaming) too, I was able to get 10~15 frames with resolution of 320x240.

CT83, 2018-04-17:
I don't know where I copied the code from but, I had used 
`        for _ in camera.capture_continuous(stream, 'jpeg'):` 
instead of
`        for _ in camera.capture_continuous(stream, 'jpeg', use_video_port=True):`
That was the source of the problem!! Getting smooth 25-30 FPS!

closed 2018-04-17 by CT83