hamuchiwa/AutoRCCar #103: How can I do this project without Raspberry Pi ?

sinanb41, 2018-04-16:
How can i do this project with ip webcam android app for streaming video and  ultrasonic sensor read and motor control from arduino ?

sinanb41, 2018-04-17:
**My collect_training_data** 
it's working but only captures frames and stuck at Start collecting images... When i press keys happens nothing and  no print but rc_control_test working nice.
      
           import numpy as np
           import cv2
           import serial
           import pygame
           from pygame.locals import *
           import time
           import os
           import urllib

           url = 'http://192.168.14.68:8080/video'  # duzenlenecek
           stream = urllib.urlopen(url)
           stream_bytes = ''
           class CollectTrainingData(object):
    
    def __init__(self):

        # accept a single connection
        # self.connection = self.server_socket.accept()[0].makefile('rb')

        # connect to a seral port
        self.ser = serial.Serial('COM3', 115200, timeout=1) #duzenlenecek
        self.send_inst = True

        # create labels
        self.k = np.zeros((4, 4), 'float')
        for i in range(4):
            self.k[i, i] = 1
        self.temp_label = np.zeros((1, 4), 'float')

        pygame.init()
        self.collect_image()

    def collect_image(self):

        saved_frame = 0
        total_frame = 0

        # collect images for training
        print 'Start collecting images...'
        e1 = cv2.getTickCount()
        image_array = np.zeros((1, 38400))
        label_array = np.zeros((1, 4), 'float')

        # stream video frames one by one
        try:
            stream_bytes = ' '
            frame = 1
            while self.send_inst:
                stream_bytes += stream.read(1024)
                first = stream_bytes.find('\xff\xd8')
                last = stream_bytes.find('\xff\xd9')
                if first != -1 and last != -1:
                    jpg = stream_bytes[first:last + 2]
                    stream_bytes = stream_bytes[last + 2:]
                    image = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8), cv2.IMREAD_GRAYSCALE)
                    
                    # select lower half of the image
                    roi = image[120:240, :]
                    
                    # save streamed images
                    cv2.imwrite('training_images/frame{:>05}.jpg'.format(frame), image)
                    
                    #cv2.imshow('roi_image', roi)
                    cv2.imshow('image', image)
                    
                    # reshape the roi image into one row array
                    temp_array = roi.reshape(1, 38400).astype(np.float32)
[... 75 lines cut ...]
            file_name = str(int(time.time()))
            directory = "training_data"
            if not os.path.exists(directory):
                os.makedirs(directory)
            try:    
                np.savez(directory + '/' + file_name + '.npz', train=train, train_labels=train_labels)
            except IOError as e:
                print(e)

            e2 = cv2.getTickCount()
            # calculate streaming duration
            time0 = (e2 - e1) / cv2.getTickFrequency()
            print 'Streaming duration:', time0

            print(train.shape)
            print(train_labels.shape)
            print 'Total frame:', total_frame
            print 'Saved frame:', saved_frame
            print 'Dropped frame', total_frame - saved_frame

        finally:
                exit(0)

           if __name__ == '__main__':
    CollectTrainingData()`

sinanb41, 2018-04-18:
pls help  @hamuchiwa

hamuchiwa, 2018-04-18:
If it doesn't output any text to the console, it means it's not in the pygame loop. Do you know at what line it hangs?

Gooose1909, 2018-04-19:
* screen = pygame.display.set_mode((300, 200))*
Add this line after pygame.init()

sinanb41, 2018-04-19:
Ok.I will try but how can i do it without rasp ?
Here is my rc_driver.py

    import threading
    import SocketServer
    import serial
    import cv2
    import numpy as np
    import math
    import urllib
           sensor_data = " " # distance data measured by ultrasonic sensor
           url = 'http://192.168.14.68:8080/video'  # duzenlenecek
           stream = urllib.urlopen(url)
           stream_bytes = ''
        class NeuralNetwork(object):

    def __init__(self):
        self.model = cv2.ml.ANN_MLP_load(mlp_xml/mlp.xml)

    def create(self):
        layer_size = np.int32([38400, 32, 4])
        self.model.create(layer_size)

    def predict(self, samples):
        ret, resp = self.model.predict(samples)
        return resp.argmax(-1)
     class RCControl(object):

    def __init__(self):
        self.serial_port = serial.Serial('COM3', 115200, timeout=1) #duzenlenecek

    def steer(self, prediction):
        if prediction == 2:
            self.serial_port.write('1')
            print("Forward")
        elif prediction == 0:
            self.serial_port.write('4')
            print("Left")
        elif prediction == 1:
            self.serial_port.write('2')
            print("Right")
        else:
            self.stop()

    def stop(self):
        self.serial_port.write('0')

    class DistanceToCamera(object):

    def __init__(self):
        # camera params
        self.alpha = 8.0 * math.pi / 180
        self.v0 = 119.907778117
        self.ay = 332.213350706

    def calculate(self, v, h, x_shift, image):
        # compute and return the distance from the target point to the camera
        d = h / math.tan(self.alpha + math.atan((v - self.v0) / self.ay))
        if d > 0:
            cv2.putText(image, "%.1fcm" % d,
                (image.shape[1] - x_shift, image.shape[0] - 20), cv2.FONT_HERSHEY_SIMPLEX, 0.7, (255, 255, 255), 2)
        return d

    class ObjectDetection(object):

    def __init__(self):
        self.red_light = False
        self.green_light = False
        self.yellow_light = False

[... 187 lines cut ...]

                    if cv2.waitKey(1) & 0xFF == ord('q'):
                        self.rc_car.stop()
                        break

            cv2.destroyAllWindows()

        finally:
            print "Connection closed on thread 1"

      class ThreadServer(object):

    def server_thread(host, port):
        server = SocketServer.TCPServer((host, port), VideoStreamHandler)
        server.serve_forever()

    def server_thread2(host, port):
        server = SocketServer.TCPServer((host, port), SensorDataHandler)
        server.serve_forever()

    # distance_thread = threading.Thread(target=server_thread2, args=('192.168.1.100', 8002))
    # distance_thread.start()

      if __name__ == '__main__':
    ThreadServer()

Gooose1909, 2018-04-19:
I am not familiar with android
Look into this ptoject
He used similar approach of using mobile device for stream

https://github.com/dps/nnrccar

sinanb41, 2018-04-19:
Android part is just video streaming it can be pc camera too.

sinanb41, 2018-04-19:
I mean it is not about android coding I will attach small old android phone and use ip webcam app for streaming' http://192.168.14.68:8080/video

sinanb41, 2018-04-19:
For car movement I will connect HC06 bluetooth module and my PC then COM port will send data

sinanb41, 2018-04-21:
@hamuchiwa  How can I do this project without Raspberry Pi ?

hamuchiwa, 2018-04-23:
@sinanb41 I assume you are asking how to do this project with an ip camera, take a look at [this](https://github.com/hamuchiwa/AutoRCCar/issues/56).

tiachachat, 2018-04-26:
@sinanb41 we are in the same boat
I changed this project a little bit
I used this app to streaming video "IP Webcam" https://play.google.com/store/apps/details?id=com.pas.webcam&hl=en
but i hava probleme with training the model
this is the code 

import numpy as np
import cv2
import time
import requests
import threading
from threading import Thread, Event, ThreadError
import serial
import pygame
from pygame.locals import *
import os
url = 'http://192.168.1.6:8080/video'

inst = True
class CollectTrainingData(object):

  def __init__(self):
    
    self.stream = requests.get(url, stream=True)
    self.thread_cancelled = False
    self.thread = Thread(target=self.run)
    print "camera initialised"
    self.thread.start()
    self.ser = serial.Serial('COM3', 115200, timeout=1)
    self.k = np.zeros((4, 4), 'float')
    for i in range(4):
        self.k[i, i] = 1
    self.temp_label = np.zeros((1, 4), 'float')
    pygame.init()
    self.inst = True
    
    
    
    
  
    
  def run(self):
    screen = pygame.display.set_mode((320,240))
    saved_frame = 0
    total_frame = 0

    # collect images for training
    print 'Start collecting images...'
    e1 = cv2.getTickCount()
    image_array = np.zeros((1, 38400))
    label_array = np.zeros((1, 4), 'float')
    
    try:
      bytes=''
      frame = 1
      while self.inst:
        
        bytes+=self.stream.raw.read(1024)
        a = bytes.find('\xff\xd8')
        b = bytes.find('\xff\xd9')
        if a!=-1 and b!=-1:
          jpg = bytes[a:b+2]
          bytes= bytes[b+2:]
          img = cv2.imdecode(np.fromstring(jpg, dtype=np.uint8),cv2.IMREAD_GRAYSCALE)
          # select lower half of the image
          roi = img[120:240, :]

          # save streamed images
          cv2.imwrite('training_images/frame{:>05}.jpg'.format(frame), img)
           
[... 90 lines cut ...]

      e2 = cv2.getTickCount()
      # calculate streaming duration
      time0 = (e2 - e1) / cv2.getTickFrequency()
      print 'Streaming duration:', time0

      print(train.shape)
      print(train_labels.shape)
      print 'Total frame:', total_frame
      print 'Saved frame:', saved_frame
      print 'Dropped frame', total_frame - saved_frame
    except ThreadError:
      self.thread_cancelled = True
        
        
  
      
    
 

  
    
if __name__ == "__main__":
  
  CollectTrainingData()

closed 2018-06-29 by hamuchiwa

VishalWrench, 2019-03-12:
## I am also trying to implement using phone camera (without rasp pi).    @sinanb41 @hamuchiwa  I have tried running this code using ip-camera android app for streaming when i press arrow keys in my keyboard it is not saving images and exiting out of the loop. when i have removed these two lines 

### image_array = np.vstack((image_array, temp_array))
### label_array = np.vstack((label_array, self.k[2])) 

### it send the data to serial and my car moves but my camera images are not saved (as i have removed those lines)
import numpy as np
import cv2
import serial
import pygame
from pygame.locals import *
import time
import os
import urllib

url = 'http://100.86.250.34:8080/video'  # duzenlenecek
stream = urllib.urlopen(url)
stream_bytes = ''
class CollectTrainingData(object):
    def __init__(self):

        # accept a single connection
        # self.connection = self.server_socket.accept()[0].makefile('rb')

        # connect to a serial port
        self.ser = serial.Serial('COM3', 115200, timeout=1) #duzenlenecek
        self.send_inst = True

        # create labels
        self.k = np.zeros((4, 4), 'float')
        for i in range(4):
            self.k[i, i] = 1
        self.temp_label = np.zeros((1, 4), 'float')

        pygame.init()
        pygame.display.set_mode((400, 400))
        self.collect_image()

    def collect_image(self):

        saved_frame = 0
        total_frame = 0
        shapeu = 320*240

        # collect images for training
        print 'Start collecting images...'
        print("Press 'q' or 'x' to finish...")
        e1 = cv2.getTickCount()
        image_array = np.empty((0, shapeu))
        label_array = np.empty((0, 4))

        # stream video frames one by one
        try:
            stream_bytes = ' '
            frame = 1
            while self.send_inst:
                stream_bytes += stream.read(1024)
                first = stream_bytes.find('\xff\xd8')
                last = stream_bytes.find('\xff\xd9')
                if first != -1 and last != -1:
                    jpg = stream_bytes[first:last + 2]
                    stream_bytes = stream_bytes[last + 2:]
                    image = cv2.imdecode(np.frombuffer(jpg, dtype=np.uint8), cv2.IMREAD_GRAYSCALE)

                    # select lower half of the image
                    height, width = image.shape
                    roi = image[int(height/2):height, :]
                    #roi = image[120:240, :]

                    #cv2.imshow('roi_image', roi)
[... 90 lines cut ...]
            try:
                np.savez(directory + '/' + file_name + '.npz', train=image_array, train_labels=label_array)
            except IOError as e:
                print(e)

            e2 = cv2.getTickCount()
            # calculate streaming duration
            time0 = (e2 - e1) / cv2.getTickFrequency()
            print 'Streaming duration:', time0

            print(image_array.shape)
            print(label_array.shape)
            print 'Total frame:', total_frame
            print 'Saved frame:', saved_frame
            print 'Dropped frame', total_frame - saved_frame

        finally:
            exit(0)

if __name__ == '__main__':
    CollectTrainingData()
    
### Here is the screenshot of the output.When i press forward key it prints forward and doesn't save image and exits from the loop and when i press backward key it send signal to the car it moves backward as i have removed the array coversion lines. help me with this plzzz.
### Thanks in advance,  Waiting for your reply
[image]