hamuchiwa/AutoRCCar #109: How to control RC car Speed ?

marcusvn, 2018-05-22:
My Car is really fast, and It can't control while i have trained. How I can decrease speed of auto RC car ? What is your solution? Can you describe detail for help me plz ?

hamuchiwa, 2018-05-24:
我当时是加了一个电位器，不过效果一般，有条件的话还是找一个速度可调节的遥控车来的更容易一些。

marcusvn, 2018-05-25:
How can I connect potentiometer with motor of RC Car ?

hamuchiwa, 2018-05-29:
I just put a potentiometer between motor and incoming wire.

marcusvn, 2018-05-30:
What potentiometer did you use?10k or 5k or what ? and Can you share for me for circuit ? or website did you use ?thanks for helping !

hamuchiwa, 2018-05-30:
I don't really remember, my rc car was disassembled long long time ago, but pick the smaller one. You can just connect one potentiometer pin to the motor, and connect another pin to the wire that originally connected to the motor. 
                                    ↓
     _____ -  M + _____www ______

RajdeepMondal, 2018-06-05:
Why dont u use PWM?

marcusvn, 2018-06-05:
How I can use PWM ? @RajdeepMondal

RajdeepMondal, 2018-06-06:
If you are using a motor driver directly supply a PWM into the enable pin
of the motor driver...

closed 2018-06-07 by marcusvn