hamuchiwa/AutoRCCar #110: Add a USB camera instead of picamera

hsharma27, 2018-05-24:
I am getting many hardware issues from pi camera and I was wondering if we can use USB camera instead of pi camera? Can anyone please suggest changes in code ? Thanks in advance.

closed 2018-05-27 by hsharma27

astirox94, 2018-09-02:
any updates on this ?

shahshravan007, 2019-01-17:
I also wanted that one

denisrexhepi, 2020-05-17:
Anyone found a solution on this?

krukhlis, 2020-05-17:
Guys, switching camera from PI Camera to USB Camera has no relations to this specific project. There are tons of guides( including videos) how to do this and what has to be adjusted in your code to use USB Camera instead of CSI. In order to make changes all you need to do is to spend ~30 mins in Google and IDE.
P.S. From my experience -- you should be careful, because not all USB Web Cams support all useful color modes and resolutions. Also, you should understand that using USB Camera will have higher latency than CSI and will be impacted by another connected USB devices, especially if your RPi is below RPi4 grade.

krukhlis, 2020-05-17:
Just 1 min googling for you:
https://www.raspberrypi.org/documentation/usage/webcams/
any here it is with OpenCV code:
https://www.pyimagesearch.com/2016/01/18/multiple-cameras-with-the-raspberry-pi-and-opencv/

This is more than enough to add support for WebCam to this project.