hamuchiwa/AutoRCCar #112: Training Accuracy Huge Problem!!!!

Cagriyoruk, 2018-05-29:
Hi,

I am having a problem with the training.No matter what I tried (Tried all the things in the issues) still couldn't get the Accuracy over %60 percent. I tried to improve the Accuracy but instead of getting good results, my accuracy fell over %40 percent. It's been 3 weeks, but I couldn't achieve a solution. I will be very happy if you can give me some tips. I built a circle track for the training. I also tried other combinations just left and just right but no result. I drove the car like 35 times for getting good results.35 other npz files but the accuracy was %30 percent. I don't know what I'll have to do. Thank you for now.

Cagriyoruk, 2018-05-30:
Hi again @hamuchiwa ,

I tried a lot of combinations.My train accuracy is overall (80-100) but the test accuracy never got over %40 percent.Is there anyone who can help me with this issue? I don't know what im doing wrong.

hamuchiwa, 2018-05-30:
@Cagriyoruk Training a model is never an easy task. You could try collecting high quality data, or maybe tuning the model (eg. add extra layers, more nodes, etc.) If you have a lot of data, then you may try changing the train/test split, allocate more data to train portion. If your track doesn't change at all, you could just eliminate test portion and using all data for training to get a high score.

closed 2018-06-07 by hamuchiwa