hamuchiwa/AutoRCCar #118: OpenCV on Mac only allows rendering in main thread

hamuchiwa, 2018-08-10:
@Moha-alfridi I got same error as yours on my mac, I did some search online, it seems OpenCV can only run in main thread. So it works if not creating a separate thread for video streaming, but the sensor thread shows connection refused. I'm opening this new issue for further discussion.

hamuchiwa, 2018-08-10:
@Moha-alfridi Found a way to work around this, `rc_driver.py` has been updated and it's working on my mac, though I haven't found a way to start OpenCV in a separate thread. The sensor thread needs to be started first, then OpenCV in main thread.

Moha-alfridi, 2018-08-10:
Thank you @hamuchiwa I will test it now ..

Moha-alfridi, 2018-08-10:
@hamuchiwa It did work 👍 👍 👍 👍 👍 
it ran into the wall but it did work though .. I believe I will start make a real track to test it around.

hamuchiwa, 2018-08-10:
@Moha-alfridi Great, best of luck :smile:

closed 2018-08-10 by hamuchiwa