hamuchiwa/AutoRCCar #122: Very Urgent Issue with rc_driver.py

IbrahimOjeili, 2018-10-06:
Im having a problem where when i run rc_driver.py it displays 5 readings from the ultrasonic sensor and then i get a broken pipe error from the raspberry pi. Please help as this is very urgent!

IbrahimOjeili, 2018-10-06:
__author__ = 'zhengwang'

import sys
import threading
import socketserver
import numpy as np

from model import NeuralNetwork
from rc_driver_helper import *

# distance data measured by ultrasonic sensor
sensor_data = None

class SensorDataHandler(socketserver.BaseRequestHandler):

    data = " "

    def handle(self):
        global sensor_data
        while self.data:
            self.data = self.request.recv(1024)
            sensor_data = round(float(self.data), 1)
            # print "{} sent:".format(self.client_address[0])
            print(sensor_data)

class VideoStreamHandler(socketserver.StreamRequestHandler):

    # h1: stop sign, measured manually
    # h2: traffic light, measured manually
    h1 = 5.5  # cm
    h2 = 5.5

    # load trained neural network
    nn = NeuralNetwork()
    nn.load_model("saved_model/nn_model.xml")

    obj_detection = ObjectDetection()
    rc_car = RCControl('COM3')

    # cascade classifiers
    stop_cascade = cv2.CascadeClassifier("cascade_xml/stop_sign.xml")
    light_cascade = cv2.CascadeClassifier("cascade_xml/traffic_light.xml")

    d_to_camera = DistanceToCamera()
    d_stop_sign = 25
    d_light = 25

    stop_start = 0  # start time when stop at the stop sign
    stop_finish = 0
    stop_time = 0
    drive_time_after_stop = 0

    def handle(self):

        global sensor_data
        stream_bytes = b' '
        stop_flag = False
        stop_sign_active = True

        try:
            # stream video frames one by one
            while True:
                stream_bytes += self.rfile.read(1024)
                first = stream_bytes.find(b'\xff\xd8')
                last = stream_bytes.find(b'\xff\xd9')
                if first != -1 and last != -1:
                    jpg = stream_bytes[first:last + 2]
                    stream_bytes = stream_bytes[last + 2:]
                    gray = cv2.imdecode(np.frombuffer(jpg, dtype=np.uint8), cv2.IMREAD_GRAYSCALE)
                    image = cv2.imdecode(np.frombuffer(jpg, dtype=np.uint8), cv2.IMREAD_COLOR)
[... 85 lines cut ...]
class Server(object):
    def __init__(self, host, port1, port2):
        self.host = host
        self.port1 = port1
        self.port2 = port2

    def video_stream(self, host, port):
        s = socketserver.TCPServer((host, port), VideoStreamHandler)
        s.serve_forever()

    def sensor_stream(self, host, port):
        s = socketserver.TCPServer((host, port), SensorDataHandler)
        s.serve_forever()

    def start(self):
        sensor_thread = threading.Thread(target=self.sensor_stream, args=(self.host, self.port2))
        sensor_thread.daemon = True
        sensor_thread.start()
        self.video_stream(self.host, self.port1)

if __name__ == '__main__':
    h, p1, p2 = "192.168.0.49", 110, 112

    ts = Server(h, p1, p2)
    ts.start()

hamuchiwa, 2018-10-09:
Hi, what's the error on your pc?

IbrahimOjeili, 2018-10-09:
Here is the error I am getting on my PuTTY client

pi@AutoRCPi:~ $ Distance : 129.8 cm
<_io.BufferedWriter name=4>
Distance : 129.7 cm
Distance : 129.8 cm
Distance : 129.2 cm
Distance : 129.7 cm
Distance : 130.2 cm
Distance : 130.1 cm
Distance : 129.3 cm
Distance : 129.1 cm
Distance : 129.3 cm
Distance : 130.1 cm
Distance : 129.2 cm
Distance : 129.7 cm
Sent 13 images in 3 seconds at 3.57fps
Traceback (most recent call last):
  File "stream_client_fast.py", line 44, in <module>
    connection.write(struct.pack('<L', 0))
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 758, in __exit_                                                                                                                                                             _
    self.close()
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 737, in close
    self.stop_recording(splitter_port=port)
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1196, in stop_r                                                                                                                                                             ecording
    self.wait_recording(0, splitter_port)
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1167, in wait_r                                                                                                                                                             ecording
    encoder.wait(timeout)
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 398, in wait
    raise self.exception
  File "stream_client_fast.py", line 40, in <module>
    camera.wait_recording(6000)
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 1167, in wait_r                                                                                                                                                             ecording
    encoder.wait(timeout)
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 398, in wait
    raise self.exception
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 267, in _call                                                                                                                                                             back
    stop = self._callback_write(buf)
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 921, in _call                                                                                                                                                             back_write
    return super(PiVideoEncoder, self)._callback_write(buf, key)
  File "/usr/lib/python3/dist-packages/picamera/encoders.py", line 298, in _call                                                                                                                                                             back_write
    written = output.write(buf.data)
  File "stream_client_fast.py", line 23, in write
    self.connection.write(self.stream.read(size))
  File "/usr/lib/python3.5/socket.py", line 594, in write
    return self._sock.send(b)
BrokenPipeError: [Errno 32] Broken pipe
Distance : 129.7 cm
Traceback (most recent call last):
  File "ultrasonic_client.py", line 53, in <module>
    client_socket.sendall(str(distance).encode('utf-8'))
ConnectionResetError: [Errno 104] Connection reset by peer

IbrahimOjeili, 2018-10-09:
Something is causing a timeout as soon as the camera starts recording

hamuchiwa, 2018-10-09:
Did you make any changes to the file?

IbrahimOjeili, 2018-10-09:
The rc_driver? no. No changes except for ports for arduino ultrasonic and stream since I am using windows

hamuchiwa, 2018-10-10:
Not sure why, I only tested this project on Mac and Linux. Did you combined two error messages together in your previous comment?

closed 2019-01-07 by hamuchiwa

Yoganandhan01, 2020-03-17:
I am also having the same problem in windows os

Yoganandhan01, 2020-03-17:
hamuchiwa, you are used GPIO.setmode(GPIO.BCM) in ultrasonic_client.py, in that I did not get any output.  Instead of I am using GPIO.setmode(GPIO.BOARD) i am getting output.
but i makes the broken pipe error