hamuchiwa/AutoRCCar #124: Hardwares to use.

supaabzz, 2018-11-17:
Hi, I'd like to try it myself with the same hardwares as you've used. I can't seem to find see anywhere what hardwares are used. 

I'm assuming raspberry pi with latest upgraded version would be better option, but would it work with pi 0 w? 
Also what other components do I need to get started.

hamuchiwa, 2018-11-30:
@supaabzz @edPratt Actually raspberry pi just sits on top of the car, there is no wiring between the car and pi. For simplicity, you can even choose not to connect an ultrasonic sensor to the pi. Arduino is connected to the remote controller directly. Since controller may vary, it's hard to say there is a standard way to set up.

AnoRebel, 2018-12-02:
Hi, i would also really like to try this and hopefully advance it more, but i cant find hardware specific anywhere or a simple tutorial describing the wiring, or a circuit diagram, could you please help me out?
I would be very thankful..

Vishwasnayak, 2019-02-27:
Whether arduino is necessary to capture images or is there any other method to capture image while training. Can anyone also tell the benefits of using arduino

hamuchiwa, 2019-02-27:
Not really a benefits, if you want to use raspberry pi to control rc car, then you need to add some extra stuff. 
If you refer how to save an image to file while collecting, you can create a folder named `training_images` and add the following line to `collect_training_data.py` after the decoding. 
```python
cv2.imwrite('training_images/frame{:>05}.jpg'.format(frame), image)
```

Vishwasnayak, 2019-02-28:
Can we capture images through mobile or any such devices instead of using pi camera. One of our faculty member suggested us to capture images in mobile is it possible? Can u please tell about this???

Vishwasnayak, 2019-02-28:
sir can you please tell me the use of using arduino. can we capture images
through camera?

hamuchiwa, 2019-02-28:
@Vishwasnayak You can use a usb camera instead of pi camera. Arduino is connected to the remote controller, it has nothing to do with video streaming in this project.

Vishwasnayak, 2019-03-01:
Thank you sir

Vishwasnayak, 2019-03-11:
Sir is there any necessary to train and test in the same environment or
room. My question is if we train(capture images) data in one room and test
it in other room, is it possible??  Thank you in advance

hamuchiwa, 2019-03-11:
@Vishwasnayak Ideally, train on one track and run on another. Train/valid split is included in the model, but it's actually better to do a k-folds cross validation in your case. In that case, make sure to collect a lot of data.

Vishwasnayak, 2019-03-12:
Thank you sir for your reply. So basically if we train lot of images then
its possible to test in any room or environment with the same track? We are
training in one room and we are supposed to demonstrate in auditorium, so
its possible no sir?   Thank you in advance

hamuchiwa, 2019-03-13:
@Vishwasnayak Yes, it's possible. Use more data and a more complex model can help.

Vishwasnayak, 2019-03-14:
Thank you sir for your reply. To use more complex data are we supposed to
change anything in the code?

hamuchiwa, 2019-03-14:
@Vishwasnayak Yes, add more layers/neurons, or if the neural network model in OpenCV doesn't work for you well, you can try to create a CNN using Tensorflow.

Vishwasnayak, 2019-03-20:
Thank you sir for your previous reply. Our raspberry pi OS is getting
corrupted once in 4 days. Do you have any particular solution for that. I
have one more question sir, when we train our car, it goes only in straight
track it doesn't take turn. So my question is how many images should we
capture to make the car take turn.

hamuchiwa, 2019-03-20:
@Vishwasnayak I'm not sure why it's corrupted, maybe try to isolate what's running one by one? Could you also direct future training related questions to #168, and perhaps upload your data? I can do some training with your data, and hope we can figure out where went wrong together.

Vishwasnayak, 2019-03-20:
Thank you sir for your response. We will send the trained data as soon as
possible

Vishwasnayak, 2019-03-22:
Sir, our car speed is quiet fast. Is it a parameter which affects the edge
detection of lane? And we also want to know is there any application or
software to read npz file. We are planning to train the car with a distance
of lane around 2 meter. So how many times we have to capture the image?
Thank you sir in advance.

hamuchiwa, 2019-03-22:
@Vishwasnayak If it's too fast, the video streaming couldn't keep up, you need to find a way to slow down the car first before collecting any data. npz fits is created by Numpy, once you start to train the model, it will call `load_data` function in `model.py`to load all npz files. To start collecting images, try to drive on entire track at least two or three times.
This video can give you an idea how fast the car runs: 
https://www.youtube.com/watch?v=KrvvmLEQk8o

ShwetaLalwani, 2019-03-22:
Hi @hamuchiwa ,when I'm trying to read . npy file images after running collect_training_data.py, I'm getting all null images. Can you tell me what must be the problem?

hamuchiwa, 2019-03-22:
@ShwetaLalwani Can you explain a bit more about null image? Are you trying to visualize the data or something else? Images will be reshaped and saved into Numpy array as npz files that will be created in `training_data` folder.

ShwetaLalwani, 2019-03-22:
@hamuchiwa yes right, I'm trying to open the npz files and look at the images that are captured ... But when I'm opening the images , they are all null (I can't see the path, full image is white). Can you plz help me with this?

hamuchiwa, 2019-03-22:
@ShwetaLalwani Once you loaded data, 
`X_train, X_valid, y_train, y_valid = load_data(input_size, data_path)`
reshape it back to its original shape, then visualize it.
```python
sample = X_train[0].reshape((120, 320)) # if using whole image, then (240, 320)
plt.imshow(sample, cmap='gray')
```

Vishwasnayak, 2019-03-23:
Thank you sir for your previous reply. If there is track with one curve and
then straight..then do we need to press the forward key again and
again(frame by frame) or continuously can we press the key till there is a
straight track?

hamuchiwa, 2019-03-25:
@Vishwasnayak Image is saved only when key is pressed, so it's more like tapping keys to move the car. Just drive on the track multiple times, or you could switch off the car and place the car on the track where you think it should be, then press arrow keys to collect while manually moving the car.

Vishwasnayak, 2019-03-25:
Thank you sir. How to read npz files. Our faculty members want to view the
images captured in npz files. How to view that sir?

Gooose1909, 2019-03-25:
from numpy import load

data = load('Numpyfilename.npz')
lst = data.files
for item in lst:
    print(item)
    print(data[item])

Use this code to print

Vishwasnayak, 2019-03-25:
thank you sir for your response. Can you suggest any methods to control
speed of car. As it is not detecting turns and even stop sign.

Gooose1909, 2019-03-25:
Actually i made a RC car on my own so i had supplied less voltage to my
motor so as to make the car run slowly
Which worked for me. Try this method i am not sure though it may work or not

Vishwasnayak, 2019-03-25:
Thank you sir. We are gibing supply of 3 v of 2 1.5 battery ..sir how did u
reduce voltage any suggestion?

Gooose1909, 2019-03-25:
In my RC car i used L298N motor driver.
So for motor it needs 12v like that
For 12v it needed 8 batteries of 1.5v
 i just used 6 instead of 8 works perfectly for one day and then u have to
recharge again.
This needed regular recharging of batteries then i used potentiometer which
made it easy for controlling RPM

Vishwasnayak, 2019-03-25:
We are doing it in a toy car of 1.2*3 volt battery which is rechargeable
one. Do u have any suggestion for this type of car to decrease its speed?

ShwetaLalwani, 2019-03-27:
@Vishwasnayak Are you able to watch the images. Can you plz show me one image? How does it look? Because when I'm opening . Npz files I can't see the images. They are very small like a dot image.

Gooose1909, 2019-03-27:
Images will be saved in different folder.

ShwetaLalwani, 2019-03-27:
Which folder?

ShwetaLalwani, 2019-03-27:
@Gooose1909 the code that you provided above to open . Npz file is showing output in vector format. So suppose for one image [36,36,37..... 115,115,136] is displayed . Is this correct?

Gooose1909, 2019-03-27:
Yes correct

ShwetaLalwani, 2019-03-27:
I mean are these values correct 36 37?
What were your values?

Vishwasnayak, 2019-03-31:
Sir, if the car speed is so fast is there any other method for perfect
training to get desired output without changing car speed

Vishwasnayak, 2019-04-09:
Sir, our pi camera lags while training . it also lags when right command is
displayed after testing. Can you provide any solution for this camera
lagging. It lags more while testing....thank you sir

Vishwasnayak, 2019-04-10:
sir could u please reply as soon as possible. Its very urgent

hamuchiwa, 2019-04-10:
@Vishwasnayak Your lag is too large to drive or it's something else...? It's common to have some delays, how many frames can you receive per second?

Vishwasnayak, 2019-04-11:
We didn't change anything on frame speed its 15 fps. Whenever car takes
left while testing pi camera hangs ...it also does not detect lane? Can you
please tell solution as quick as possible. Thank you sir

Vishwasnayak, 2019-04-12:
Please reply sir .......

Vishwasnayak, 2019-04-12:
Its too urgent. We have to submit it by saturday

hamuchiwa, 2019-04-12:
@Vishwasnayak I honestly do not have an answer for that, you are using a newer model of raspberry pi and a more powerful computer, maybe something wrong with your internet?

Vishwasnayak, 2019-04-12:
We used raspberry pi model b+ . it detect stop sign and obstacle....but it
fails to detect lane....car will go straight even if the road ends and does
not take turn? Can u please send xml file of yours?