hamuchiwa/AutoRCCar #125: Error while model_training

V1shishta, 2018-12-05:
This is the error I am encountering while running model_training.py 

```
  File "C:\Users\Vishishta\Desktop\pr\computer\model_training.py", line 8, in <module>
    X_train, X_valid, y_train, y_valid = load_data(input_size, data_path)
  File "C:\Users\Vishishta\Desktop\pr\computer\model.py", line 30, in load_data
    X = np.vstack((X, train))
  File "C:\Users\Vishishta\AppData\Local\Programs\Python\Python36-32\lib\site-packages\numpy\core\shape_base.py", line 234, in vstack
    return _nx.concatenate([atleast_2d(_m) for _m in tup], 0)
ValueError: all the input array dimensions except for the concatenation axis must match exactly
```

hamuchiwa, 2018-12-05:
It seems it's an array shape mismatch issue, double check your data shape as well as the shape of `X_train, X_valid, y_train, y_valid`.

V1shishta, 2018-12-06:
Hey hamuchiwa,

Thanks for the reply, I have found the error and got it fixed. I just changed input_size to 76800 from 320*120 and it worked.

Thanks again. Keep up the good work.

closed 2019-01-07 by hamuchiwa