hamuchiwa/AutoRCCar #126: Communication between RPi, Arduino and PC

DavidAtanasoski, 2018-12-05:
I have a few questions for this project which I didn't find anything to answer it by myself. How does RPi and Arduino communicate? How do you control the motors and do you have a wiring diagram for it? I found a couple of photos here [https://github.com/hamuchiwa/AutoRCCar/issues/124](url) , but they're not very clear. To make it simple, I'm asking you how do you send commands to the motors? From RPi or Arduino? And also where is that code? 

I hope you understand me. Thank you for advance.

hamuchiwa, 2018-12-05:
Please see answers below: 
1. They don't communicate to each other directly, there is a computer in between.
2. There is no motor control, and there is no commands sent to the motor, simply Arduino sends commands to the controller, the remote control part is untouched. 
3. For collecting data, you need to drive the car by your self via keyboard. In the video, the car drives itself.

DavidAtanasoski, 2018-12-05:
So, basically arduino sends signals to the controller and the controller is controlling the motors by radio frequencies. Am I right? And another question, how can I set up this code on my computer/raspberry?

hamuchiwa, 2018-12-05:
Yes, you are correct. Check `readmd` file for setting up project environment and how to run.

DavidAtanasoski, 2018-12-10:
It's me again. Sorry if I bother you, but I have another questions for you. I'm trying to make this code works on Windows 10, I'm using PyCharm as IDE. I can't figure how to make it work, when I run rc_driver.py I get error with the model. I can't figure what's model and how to make it work. Would you like to help me?

Thank you for advance.

DavidAtanasoski, 2018-12-10:
There you have a picture of the problem. 

[image]

hamuchiwa, 2018-12-10:
It seems you haven't trained the model yet. I would suggest trying test files in the `test` folder, in which it includes a Jupyter notebook walkthrough on how to train a model.

DavidAtanasoski, 2018-12-11:
I don't get what you're telling me. When I run **stream_server_test.py** from **test** folder I get this error
[image]

hamuchiwa, 2018-12-11:
You may need to change the host IP, modify the IP to your localhost address and give it a try.

DavidAtanasoski, 2018-12-20:
It doesn't work. I've changed to my computer IP address, but nothing is happening. What about the RPi? I've also changed there in the **stream_client.py** , but as I said nothing changed.

DavidAtanasoski, 2018-12-20:
It seems I've just solve that problem , but partly. Now I have this problem...
[image]

hamuchiwa, 2018-12-20:
@DavidAtanasoski I think that's something related to opencv installation. I only tested this project on Linux and Mac. If windows gives you too much headache, it might be easier to use a linux virtual machine.

DavidAtanasoski, 2018-12-20:
I will try ithat. Just tell me what linux were you using (Ubuntu, Manjaro ...?) And which IDE? Thank you.

hamuchiwa, 2018-12-20:
@DavidAtanasoski Ubuntu 16.04 and PyCharm

DavidAtanasoski, 2018-12-24:
Problems and more problems... Now, I have the next problem on Ubuntu. I've read on the internet that this is due to newest python version, but I'm not sure. Any ideas? This also happens in the PyCharm, not only when I try to install it by terminal.
[image]

DavidAtanasoski, 2019-01-05:
I've setup the interpreter successfully. I can run now the **rc_driver.py** from `computer `directory , but I don't have any model that's trained. Also, when I run the **stream_client.py** I get the following error.

**Note:** I have created the environment from the **environment.yml** file and it works for the **rc_driver.py** ( OpenCV works there ). The interpreter is the same for the whole project.
[image]

DavidAtanasoski, 2019-01-05:
And also how can I train my own model or how can I use yours ( if it's included in this project here ) ? How to establish connection with RPi? 

Thank you in advance.

hamuchiwa, 2019-01-07:
@DavidAtanasoski `stream_server_test.py` is simply a shrined version of `rc_driver.py`, once you started the server, start `stream_client_test.py` on raspberry pi. 
What error did you get from PyCharm, were you trying to install some package?

You need to collect image data before training your model, one you have done that (`collect_training_data.py`), if you are not familiar with model training yet, there is a jupyter notebook `train_predict_test.ipynb` that walks thought nn training in OpenCV.

DavidAtanasoski, 2019-01-07:
The error is shown on the picture. OpenCV is missing and I can't install it in PyCharm.

hamuchiwa, 2019-01-08:
@DavidAtanasoski Everything should be installed with the yml file. Make sure you select the `auto-rccar` as the project interpreter in PyCharm.

DavidAtanasoski, 2019-01-09:
@hamuchiwa When I run `collect_training_data.py` I get the following error. What should I do? I think that the problem is with the host address. How can I set up that address and where to find? Also, what serial port should I use? 

Thank you for your patience and I hope I'll make it work as soon as possible.

[image]

hamuchiwa, 2019-01-10:
@DavidAtanasoski Make sure all test files run properly before moving on, change ip address to match your localhost on both PC and raspberry pi.  It's located at the bottom of server/client file.

DavidAtanasoski, 2019-01-10:
@hamuchiwa What's my localhost? How can I check that?

DavidAtanasoski, 2019-01-10:
Do you have precise solution for this problem?

DavidAtanasoski, 2019-01-10:
A little update: I've changed the address on both PC and RPi and when I start `collect_training_data.py` on PC and after that `stream_client.py` on RPi, I get nothing. Nothing is showing, neither an error or something else...

hamuchiwa, 2019-01-11:
@DavidAtanasoski Type`ifconfig` in terminal to find your ip. If you are using VirtualBox, don't forget to set network adapter to `Bridged Adapter` in virtual machine settings. 
Also since you are on Linux, change the Arduino serial port to yours as well. The one in the code is for mac.

DavidAtanasoski, 2019-01-12:
@hamuchiwa I've solved the problem. It runs perfect. So, what's next? How can I train a model? Do i need to drive the car on a road and capture it on the RPi camera or something else? And how to use trained model in `rc_driver.py`. 

Thank you in an advance.

hamuchiwa, 2019-01-14:
@DavidAtanasoski Yes, you need to drive the car and collect data first, then train the model. Once it's trained, `rc_driver.py` will load the model and use it.

closed 2019-01-14 by hamuchiwa