hamuchiwa/AutoRCCar #127: Problem while running rc_driver.py

V1shishta, 2018-12-06:
Hey Hamuchiwa,
Thanks again for the marvelous project.

I have encountered a problem while running the file rc_driver.py
You have passed two ports in the server class like this
`if __name__ == '__main__':
    h, p1, p2 = "192.168.43.94", 22

    ts = Server(h, p1, p2)
    ts.start()
`
I cant understand what will be my p2 since p1=22 has been working for me since start except this (rc_driver.py) file.
Please reply soon, I have a presentation to give on this project day after tomorrow.

Thanks 👍

hamuchiwa, 2018-12-06:
@V1shishta You can pick two big port number as long as they are free to use, eg. 8000 and 8002 (as in project), or 10000 and 10001.

V1shishta, 2018-12-06:
Thanks for the reply. 
So any port number will work on p2 regardless of what I give on p1?
Thanks again.

hamuchiwa, 2018-12-06:
@V1shishta Correct, as long as those two ports are different and available.

V1shishta, 2018-12-07:
Hey hamuchiwa,
This is the error I am encountering while running rc_driver.py and stream_client.py

`Traceback (most recent call last):
  File "/home/pi/camstream.py", line 27, in <module>
    connection.write(stream.read())
  File "/usr/lib/python3.5/socket.py", line 594, in write
    return self._sock.send(b)
BrokenPipeError: [Errno 32] Broken pipe`

In stream_client.py , I have given port no 22 which is p1 in rc_driver.py
Please reply fast.

hamuchiwa, 2018-12-10:
@V1shishta This error is a bit unclear, what's the error on computer? Make sure start the server on computer first, then start the client on pi.

closed 2019-01-07 by hamuchiwa