hamuchiwa/AutoRCCar #128: UART communication with RPi and Arduino

sundar19, 2018-12-24:
Do I have to connect the RPi and Arduino for Uart via USB cable or using Rx and Tx between Arduino and RPi, Thank You Mr.Hamuchiwa

hamuchiwa, 2018-12-27:
Hi, there is no connection between pi and arduino. Arduino is connected to the computer.

sundar19, 2018-12-28:
Mr.hamuchiwa , 
1.Then how will pi and Arduino communicate with each other using UART ?
2.Where is the motor driver chip connected to?
3.is motor driver directly connected to pi? I
Thank you sir!

sundar19, 2018-12-28:
C:\Anaconda3\envs\auto-rccar\python.exe C:/Users/krish/PycharmProjects/auto-rccar/venv/rc_control_test.py
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "C:/Users/krish/PycharmProjects/auto-rccar/venv/rc_control_test.py", line 72, in <module>
    RCTest()
  File "C:/Users/krish/PycharmProjects/auto-rccar/venv/rc_control_test.py", line 13, in __init__
    self.ser = serial.Serial("/dev/tty.usbmodem1421", 115200, timeout=1)    # mac
  File "C:\Anaconda3\envs\auto-rccar\lib\site-packages\serial\serialwin32.py", line 31, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:\Anaconda3\envs\auto-rccar\lib\site-packages\serial\serialutil.py", line 240, in __init__
    self.open()
  File "C:\Anaconda3\envs\auto-rccar\lib\site-packages\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port '/dev/tty.usbmodem1421': FileNotFoundError(2, 'The system cannot find the path specified.', None, 3)

can i know what is this problem?

sundar19, 2018-12-28:
C:\Anaconda3\envs\auto-rccar\pythonw.exe C:/Users/krish/PycharmProjects/auto-rccar/stream_server_test.py
Traceback (most recent call last):
  File "C:\Anaconda3\envs\auto-rccar\lib\site-packages\numpy\core\__init__.py", line 16, in <module>
    from . import multiarray
ImportError: DLL load failed: The specified module could not be found.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:/Users/krish/PycharmProjects/auto-rccar/stream_server_test.py", line 3, in <module>
    import numpy as np
  File "C:\Anaconda3\envs\auto-rccar\lib\site-packages\numpy\__init__.py", line 142, in <module>
    from . import add_newdocs
  File "C:\Anaconda3\envs\auto-rccar\lib\site-packages\numpy\add_newdocs.py", line 13, in <module>
    from numpy.lib import add_newdoc
  File "C:\Anaconda3\envs\auto-rccar\lib\site-packages\numpy\lib\__init__.py", line 8, in <module>
    from .type_check import *
  File "C:\Anaconda3\envs\auto-rccar\lib\site-packages\numpy\lib\type_check.py", line 11, in <module>
    import numpy.core.numeric as _nx
  File "C:\Anaconda3\envs\auto-rccar\lib\site-packages\numpy\core\__init__.py", line 26, in <module>
    raise ImportError(msg)
ImportError: 
Importing the multiarray numpy extension module failed.  Most
likely you are trying to import a failed build of numpy.
If you're working with a numpy git repo, try `git clean -xdf` (removes all
files not under version control).  Otherwise reinstall numpy.

Original error was: DLL load failed: The specified module could not be found.

Process finished with exit code 1
numpy not working in pycharm pls help!!

sundar19, 2018-12-29:
Is it compulsory to work in a conda environment or using an idle is enough??...which version of python and opencv did you use?
Thank You

sundar19, 2018-12-31:
Thank You so much Mr.Hamuchiwa

sundar19, 2018-12-31:
Traceback (most recent call last):
  File "C:\Python27\stream_server_test.py", line 50, in <module>
    VideoStreamingTest(h, p)
  File "C:\Python27\stream_server_test.py", line 12, in __init__
    self.server_socket.bind((host, port))
  File "C:\Python27\lib\socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 10049] The requested address is not valid in its context

can you help me with this?

sundar19, 2019-01-01:
Mr.Hamuchiwa do I have any sampled data sets pretrained so that I can have an overview. Thank You!

sundar19, 2019-01-02:
File "ultrasonic_client.py", line 51 in <module>
   client_socket.send(str(distance))
socket.error: [Errno 32] Broken pipe
please help me with this and previous issue 
Thank you

hamuchiwa, 2019-01-02:
@sundar19 There is a jupyter notebook `train_predict_test.ipynb` in `test/model_train_test ` directory that walks through model training using sample data. 
If you haven't done it so, try test files in `test` folder to make sure your connection works. Also don't forget to change the host ip if yours is different.

sundar19, 2019-01-03:
File "ultrasonic_client.py", line 51 in 
client_socket.send(str(distance))
socket.error: [Errno 32] Broken pipe
please help me with this and previous issue
Thank you Mr.hamuchiwa
but Iam getting a series of sensor data values after which the connection terminates automatically 
what can I do in this regard?

sundar19, 2019-01-03:
Thank You so much Mr.hamuchiwa

sundar19, 2019-01-04:
Mr.hamuchiwa Iam not going to use a remote control and will use only keyboard .How do I drive the car for training data?Can I directly connect the rpi to DC motor driver IC
Thank You once again!

hamuchiwa, 2019-01-04:
@sundar19 You need to drive the car with keyboard, whenever there is a key press action, image data is saved along with its label. Take a look #92 if you plan to use raspberry pi solely.

sundar19, 2019-01-05:
Mr.Hamuchiwa I now understand that computer and Arduino are connected to communicate via UART (USB cable) and the arduino is connected (by soldering) to the remote of the car.The direction buttons of remote are replaced with signal from Arduino. Please  reply if this is correct and also please send me the link of the RC car you bought.

hamuchiwa, 2019-01-07:
@sundar19 Yes, it's correct . Although I think it would be a much better option by control rc car from raspberry pi directly. I bought the rc car on Amazon, the link doesn't exist anymore. You may need to improvise on this.

closed 2019-01-07 by hamuchiwa

sundar19, 2019-01-09:
Thank You so much sir