hamuchiwa/AutoRCCar #129: Mobile setup without Arduino

scud3r1a, 2019-01-03:
Hey, since the Pi is able to send digital signals via GPIO pins, why shouldn't one use this interface to access the remote controller (or in a further step the engine and steering controller itself). 

In either of these cases, the Raspberry Pi would just need a WiFi connection to transmit image data and accept control commands for the radio controller (or the engine/steering controller).

What about the idea to abandon the usage of an Arduino?

maelh, 2019-01-03:
You can't access the remote control with the Pi, since the remote and the car (and therefore the Pi) are in different physical locations.
You could access the receiver controller in the car, or better yet, directly control the motors/actuators. But it's more complicated and requires disassembling the car, routing the cables so they don't get in the way/drill holes in the car's cover etc.
It's also very likely going to make the remote unusable, since you'll interfere with the receiver.

scud3r1a, 2019-01-03:
Last thing is an issue, you got a point there.

But still, I pointed out the possibility to place the remote controller on the car as well. Then, Pi and radio controller where close to each other. Still not an opportunity?

sundar19, 2019-01-04:
How do we drive the car to train our data?should rpi be directly connected to motor driver IC?

scud3r1a, 2019-01-04:
Doesn't have to be. When the Pi and the remote controller are connected, controlling the car (transporting both the Pi and its own radio controller on the roof) is made possible.

As I said, the Pi has to be accessible via Wi-Fi and so, the keyboard commands can be given through that way of transmission to the Pi which forwards them to the radio controller via GPIO.

sundar19, 2019-01-04:
RC and wifi are entirely different channels?...how does our keyboard commands control gpio of rpi?

scud3r1a, 2019-01-04:
Nevermind, I think I know what I am about to try and the only problem I see is that possible realtime response delay...

closed 2019-01-04 by scud3r1a

maelh, 2019-01-06:
Yeah, playing around is part of the fun, so just try different solutions.
My experience was that the pi and a battery already put quite some weight on it. So you may be more successful if you just remove the car cover, and interface with the car directly, removing the receiver and remote control.

You could still control the car over wifi (and the pi), and drive it over the computer's keyboard.