hamuchiwa/AutoRCCar #132: Ultrasonic test

kalmastenitin, 2019-01-08:
I am trying to run Ultrasonic server test & ultrasonic client - I m getting Errors 
on server (streaming sensor_data = float(self.connection.recv(1024))
ValueError: could not convert string to float)
on client (TypeError: a bytes-like object is required, not 'str')

all things set up properly

hamuchiwa, 2019-01-08:
@kalmastenitin Python3 and Python2 handle `str` differently. What's the default python on Raspbian?
For using python3 on raspberry pi,  you can change`ultrasonic_client.py` line 52 to `client_socket.send(str(distance).encode('utf-8))`

kalmastenitin, 2019-01-09:
it worked for me now i am using python3.7 on server and client side. Thanks

closed 2019-01-09 by kalmastenitin