hamuchiwa/AutoRCCar #138: Overview of the Project

SaiSurya9999, 2019-01-16:
I have few doubts about assembling..??
-- First one Should we connect audrino to Raspberry pie..?? (How to connect those two.??)
-- Raspberry pie and audrino will be there inside the car..??
-- Ultrasonic sensor and Camera module should be connected to Raspberry pie..??
-- Connection between raspberry pie and PC will be through sockets or any wifi module..??(Please explain the communication b/w them)
-- What about calibaration of camera should we do calibiration first and then attach the pie on the car..??
-- I have uploaded an image which i drew is it correct what i understood there..??
---------------------Thank you for your constant support-------------------------------
[image]

SaiSurya9999, 2019-01-16:
[image]
I have purchased this car from amazon...Did you use the same car....

hamuchiwa, 2019-01-16:
Take a look #124. 
1. Arduino is not connected to Raspberry pi, it's connected to the remote controller. 
2. You can give it a try to put pi in side...
3. Yes.
4. Wifi, so they need to be in the same network. 
5. If you are using pi camera module, you don't need calibration. 
6. It looks similar, maybe not much difference.

SaiSurya9999, 2019-01-17:
Another one should i create the same environment in rasbian os i mean in the pi..?? How to start server in pc and client in the pi could you please provide the command lines to do that..??
Did you use rasbian os in pi..?? 
---------Thank you-------------

tejachotu, 2019-01-24:
How to interface arduion with raspberry Pi??

SaiSurya9999, 2019-01-31:
They are not connected @tejachotu  they are seperate uno will be connected to PC via usb and pi will be fixed on the car two different modules here..

tejachotu, 2019-02-02:
How to create and use miniconda environment

tejachotu, 2019-02-02:
Can you mail me about miniconda environment

tejachotu, 2019-02-02:
Can you explain me how to run codes and how to use miniconda environment

hamuchiwa, 2019-02-07:
@tejachotu Refer to [conda doc](https://conda.io/projects/conda/en/latest/user-guide/tasks/manage-environments.html) on how to mange conda environment.

closed 2019-02-07 by hamuchiwa

kxrf97, 2019-02-23:
@SaiSurya9999 what i libray i need to install in my Pi? and also did i need to create the environment in my pi?

kxrf97, 2019-02-23:
@hamuchiwa 可以帮帮我吗，我的Pi 需要下载什么library 还是软件吗

kxrf97, 2019-02-23:
@hamuchiwa 还是只是把 raspberry pi 的 文件移到 Pi 来运行就行了？

hamuchiwa, 2019-02-23:
@kxrf97 不需要安装什么，系统都带了。直接把文件复制过去，再改一下ip地址就可以了。

kxrf97, 2019-02-24:
@hamuchiwa 那你使用的是 OS 是 NOOB 还是 raspbian strech 在 pi 上, 暂停 和 红绿灯 可以直接用 作者提供的 xml file 吗？ 还是我得重新 训练过

hamuchiwa, 2019-02-24:
@kxrf97 那两个其实是一样的，你就从官方下载NOOBS，然后选择安装raspbian。Stop sign可以直接用，红绿灯除非你能找着一样的，要不然还是自己拍点儿图片训练一下吧。