hamuchiwa/AutoRCCar #139: To be continued

DavidAtanasoski, 2019-01-16:
@hamuchiwa Because you closed the last issue ( I don't know why ), I have more questions about you. So, how can I train the model and use it? In your video on [Youtube ](https://www.youtube.com/watch?v=BBwEF6WBUQs) I can see that you use different files than these which are uploaded here. Can you explain me how can I make it drive this car, please? Does this car recognize drawn road or just road from paper? 

Sorry If I bother you, but I'm very enthusiastic for this project and I'm trying to make it work.

Sincerely,
David Atanasoski

hamuchiwa, 2019-01-16:
Hi, you can still add comment to a closed thread. If all test files work for you, then you can move on. To train a model, follow the steps in readme, you can collect data multiple times, each time a file is saved. The model only works with what it 'sees' in training data, if you want your model works for different road pattern, then you need to collect data for each pattern.

DavidAtanasoski, 2019-01-16:
@hamuchiwa Basically, I'm collecting data with `collect_training_data.py` right? Do I have to change the name of the data I load in the code? Now, I think it stands "saved_model/nn_model.xml", do I have to change this? And what is `nn_model.xml` ?

hamuchiwa, 2019-01-16:
No, data file will be named as timestamp when loading, it just loads all files in that folder.
No either, model file will be saved in that folder. `nn_model.xml` is your neural network model.

hamuchiwa, 2019-01-16:
No, `model_training.py` will create `saved_model` folder. 
Yes.
It depends on data size and your computer power, but I think you can start with collecting one or two hundreds good quality images.

DavidAtanasoski, 2019-01-16:
@hamuchiwa Thanks for your responding to my problem. I will work on this and if I have any problems I'll come back.

Thank you again.

closed 2019-02-07 by hamuchiwa

DavidAtanasoski, 2019-02-12:
I have training problem. I can't achieve high trained percentage. I don't know the reason. How should I train the car properly? It's very stressful.

Thanks for your help.

DavidAtanasoski, 2019-02-12:
And also how high should be the camera?

hamuchiwa, 2019-02-13:
The accuracy highly depends on your data quality. A couple things you can do: you can manually put the car (not running) on the track then collect data, in case of manually driving does not generate good data; tune your model; use CNN (opencv version or Tensorflow) instead of basic neural network.

I think angle matters more than height. It seems better if track fills most of the image (100% is better if you could do that), compared to just slice half the image.

DavidAtanasoski, 2019-02-13:
Does it mean that I have to change the code  to use CNN? Also, how this code works? I mean how it makes a prediction and turns like it should. I'm trying to figure it out. In my case I'm trying to drive the car using the keys on the keyboard and it works. But, the problem is data, it doesn't achieve a high accuracy. Do I have to change the path? To use black paper?

I need help for this. Thank you in advance.

DavidAtanasoski, 2019-02-13:
Or maybe to change the camera resolution? And how?

hamuchiwa, 2019-02-13:
Yes, you need to make some changes to use a different model. Good quality data means you label it mostly correct.

As for resolution, 320x240 works fine, you can give it a try if using a higher resolution causes any delays or not. Resolution can be set in `stream_client.py`, also don't forget to modify the image input size on server side code. 

If you haven't seen this yet, take a look at this video make by "aha124226675" https://www.youtube.com/watch?v=KrvvmLEQk8o

DavidAtanasoski, 2019-02-13:
I've seen that video and I'm impressed. As I said I'm working to make mine as good as this, but the accuracy is a problem. What should I change in the code to be more effective? Every help is needed.

DavidAtanasoski, 2019-02-14:
@hamuchiwa The current resolution is `120 x 320` right ?

DavidAtanasoski, 2019-02-14:
And how much the "saved frames" should be?  I assume that they're needed to create the model?

hamuchiwa, 2019-02-15:
Yes, it's 240x320 from raspberry pi then halved to 120x320. Since the track pattern is very simple, roughly two hundred images should be sufficient. The images and labels will be saved as a npz file after collecting, but you could also save jpeg frames as you wish.

DavidAtanasoski, 2019-02-15:
With 200 images saved in .npz file I can't achieve enough accuracy to drive the car. Would you like to help me to fix this problem?

DavidAtanasoski, 2019-02-15:
I've tried different paths, but unfortunately, nothing changed.

hamuchiwa, 2019-02-15:
You could try to use full image 240x320 and tilt down the camera angle. Collect more data, balance the data and tune the model can also help.

DavidAtanasoski, 2019-02-15:
@hamuchiwa How can I do that? How to tune the model and balance the data?

hamuchiwa, 2019-02-17:
Model tuning is never an easy task, you could add more layers or hidden units. If you are not familiar with neural network in general, I would recommend you go through some articles/tutorials. Nowadays, convolutional neural network is widely used for computer vision problems, instead of the simple neural network in this project. If you decide to go that path, check out Tensorflow/Keras. 

Balance data means for each label (forward, left and right), the number of images are roughly equal. I assume you collected more forward images than the others, in this case, collect more left and right images.

DavidAtanasoski, 2019-02-17:
@hamuchiwa I don't collect images for left, right and forward  I just drive the car on a path and collect data. That's what I'm asking you all the time, how should I train the car? I need details in order to make it work.
Thanks.

hamuchiwa, 2019-02-17:
If you have collected data, just run `model_training.py`.

DavidAtanasoski, 2019-02-17:
I did that. But the accuracy is very low.

DavidAtanasoski, 2019-02-17:
I'm trying to make it drive on a simple path ( just forward ), but it doesn't want to go. Also, I've tried on a simple right turn, it doesn't work either. @hamuchiwa

DavidAtanasoski, 2019-02-17:
Should I collect images for all of them (forward, right, left) directions in one .npz file or they should be in a different?

hamuchiwa, 2019-02-17:
A simple straight line should work, since you labeled all images with `forward`, it should predict `forward` too. Can you share your one or two training images here?

DavidAtanasoski, 2019-02-17:
@hamuchiwa 
[image]
[image]

DavidAtanasoski, 2019-02-17:
I've also narrowed the path but same result.

DavidAtanasoski, 2019-02-17:
When I drive the car on this straight path and then train the the model with `model_training.py` I get low accuracy.

hamuchiwa, 2019-02-17:
Can you post images captured by the raspberry pi camera?

DavidAtanasoski, 2019-02-17:
How can I see the images when they're in .npz file?

DavidAtanasoski, 2019-02-18:
@hamuchiwa 

[image]

hamuchiwa, 2019-02-19:
Image looks fine, so what's your accuracy based on this data?

DavidAtanasoski, 2019-02-19:
Below 50%

hamuchiwa, 2019-02-19:
Could you upload your data or share a link to your data?

DavidAtanasoski, 2019-02-21:
Unfortunately not because I've deleted the data.

hamuchiwa, 2019-02-21:
Okay, can you collect about 150 images on straight line?

DavidAtanasoski, 2019-02-21:
Just to drive once or multiple times?

hamuchiwa, 2019-02-21:
You can drive, or do this: just put the car on track without switching on, then move the car forward a bit manually many times as you press the arrow key. You could do once or multiple times.

kxrf97, 2019-03-06:
@DavidAtanasoski did you manage tackle the problem? i have same problem as yours

DavidAtanasoski, 2019-03-06:
Unfortunately not. I didn't manage to fix it.

kxrf97, 2019-03-07:
@DavidAtanasoski  even i achieve high accuracy my car wont perform what i expected

somalwar, 2019-03-17:
same problem here. Does not drive as expected. Bad prediction results