hamuchiwa/AutoRCCar #140: Connection issues

shahshravan007, 2019-01-17:
1)Which raspberry Pi is needed low cost
2)Can I connect rc to raspberry Pi so it will be be efficient
3)How to connect from WiFi
4) please help me with that can you give your email address 
My email address is shahshravan007@gmail.com
Thanks

SaiSurya9999, 2019-01-17:
I think raspberry pi 3 model b+ should be used..!!

geek1111, 2019-01-17:
@shahshravan007 
1)Raspberry pi 3b/3b+ (recommended)
2)Didn't understand your question.
3)Follow steps mentioned in [README.md](https://github.com/hamuchiwa/AutoRCCar/blob/master/README.md)
4)No, It'll be better if you ask your queries / problems over here many people active here, also if someone else is facing similar issues and if anybody comes with solution we can keep this as a reference for others.

shahshravan007, 2019-01-17:
@geek1111
Can I use raspberry Pi at the place of aurdino

SaiSurya9999, 2019-01-31:
No dude sorry for the late reply you cannot use pi in place of audrino why because they are seperate uno is used to control remote and pi will be on car sending sensor data to computer

shahshravan007, 2019-01-31:
Sorry sir but I wanted to ask that can I connect remote to raspberry Pi without using uno

shahshravan007, 2019-02-10:
Hey @saisurya9999 please give me your any contact I want do this project please help me me with that 
Please give reply immediately please

shahshravan007, 2019-02-10:
Hey @saisurya9999 please contact me please
My mail id is shahshravan007@gmail.com so please
I will give your charges so please contact me