hamuchiwa/AutoRCCar #147: training dataset

sundar19, 2019-01-24:
from model import NeuralNetwork
this module is not found , what can i do?
thank you in advance

sundar19, 2019-02-03:
Mr.Hamuchiwa,
Traceback (most recent call last):
  File "D:/Python36/model_training.py", line 8, in <module>
    X_train, X_valid, y_train, y_valid = load_data(input_size, data_path)
  File "D:/Python36\model.py", line 31, in load_data
    X = np.vstack((X, train))
  File "D:\Python36\lib\site-packages\numpy\core\shape_base.py", line 234, in vstack
    return _nx.concatenate([atleast_2d(_m) for _m in tup], 0)
ValueError: all the input array dimensions except for the concatenation axis must match exactly
I get this error while running  model_training.py 
what shall i do?
thank you in advance

hamuchiwa, 2019-02-07:
@sundar19 Check your numpy array shape and data shape, make sure they match.

closed 2019-02-07 by hamuchiwa