hamuchiwa/AutoRCCar #150: Problem streaming in rc_driver.py

sundar19, 2019-02-08:
Exception happened during processing of request from ('192.168.0.2', 39178)
Traceback (most recent call last):
  File "C:\Python27\lib\SocketServer.py", line 290, in _handle_request_noblock
    self.process_request(request, client_address)
  File "C:\Python27\lib\SocketServer.py", line 318, in process_request
    self.finish_request(request, client_address)
  File "C:\Python27\lib\SocketServer.py", line 331, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "C:\Python27\lib\SocketServer.py", line 652, in __init__
    self.handle()
  File "C:\Python27\rc_driver.py", line 157, in handle
    sys.exit()
SystemExit

Mr.Hamuchiwa,
 I found that ServerSocket gets the ultrasonic data but not the videostreaming data,while this works in stream_server_test.py program,please respond
      Thank You

sundar19, 2019-02-08:
And why do you use serversocket instead only socket is enough?

hamuchiwa, 2019-02-13:
No particular reason, it just simplifies things a bit.

closed 2019-02-13 by hamuchiwa