hamuchiwa/AutoRCCar #151: Training

SaiSurya9999, 2019-02-15:
Does training stores all the video frames and action we took i mean forward or left or right.
Where will they be stored..??
Thank you..!!

hamuchiwa, 2019-02-15:
Image along with label are saved only when there is a key press action. In the end, data is saved as a npz file in `training_data` directory.

SaiSurya9999, 2019-02-17:
Thank you @hamuchiwa ..!!

SaiSurya9999, 2019-02-17:
In the model_training folder we are having an npz file is it the data you have trained can we directly use model without training as the trained doc i mean npz  already present..?? (By creating the same environment as shown in the video)

hamuchiwa, 2019-02-17:
There is no pre-trained model. If you refer to the data file in `model_train_test` folder, it's just a example to walk you through how to use neural network in opencv.

SaiSurya9999, 2019-02-20:
ohh ok Thank you @hamuchiwa..!!

Vaibhav18t, 2019-02-22:
Sir while trying to run the training model using collect_training_data.py it runs on the Anaconda prompt.
Then when I am trying to run the stream_client.py on the raspberry pi, the Anaconda running Server is getting terminated showing the following error. Please tell me how to clear this error. It is Urget. I hope to hear from you soon.
[image]

hamuchiwa, 2019-02-22:
@Vaibhav18t Duplicated? #153

Vaibhav18t, 2019-02-23:
Yes Sir. The code is the same. Made some changes in the IP address and port number, and as some changes in the car's arduino sketch for controls. Could you please tell me how to re-solve this issue??@hamuchiwa #153

hamuchiwa, 2019-02-23:
@Vaibhav18t As mentioned in #153, find your arduino port on your windows computer, then replace the one in the code.

Vaibhav18t, 2019-02-23:
Sir, which module should I use to connect my arduino to my pc wirelessly. Also I thank you for your help sir.
@hamuchiwa

Vaibhav18t, 2019-02-23:
Sir how should I connect the arduino to my pc and at the same time to the car? Should I use a bluetooth module to connect to the arduino from my pc?

hamuchiwa, 2019-02-23:
@Vaibhav18t Arduino is connected to the remote controller not the car, see #114.

Vaibhav18t, 2019-02-24:
@hamuchiwa Sir we have made a motor car instead of the rc-car and we thought we would interface with the arduino with a bluetooth module with our pc. I want to ask you that if we give the bluetooth address instead of usb port address, will it work??

closed 2019-03-27 by hamuchiwa