hamuchiwa/AutoRCCar #154: How to find the forward, reverse, right and left pins of controller for arduino?

MyFirstRoboticProject, 2019-02-23:
Do I have to apply some voltage source? My controller uses two AA batteries

MyFirstRoboticProject, 2019-02-23:
By controller I mean the radio controller for the RC car

hamuchiwa, 2019-02-23:
It depends on your controller, you may need to do some trials and errors. This project uses two AA batteries which is not a desirable design, it's better powered by Arduino, see #124, #88.

MyFirstRoboticProject, 2019-02-23:
Thank you for the reply. Since this is my first time working on robotics, I'd like to know a bit about the way of how to do these trials and errors for finding the four pins defining forward, reverse, etc. on the controller circuit board.

hamuchiwa, 2019-02-23:
General rule is to follow the connection between a button and which pin it's connected to on the board.  Good luck!

MyFirstRoboticProject, 2019-02-24:
Finally was able to solve my problem by powering the controller from arduino rather than the batteries. Transmitter used TX-2B IC so the schematics in the Web helped a lot. Thanks for helping!

closed 2019-02-24 by MyFirstRoboticProject