hamuchiwa/AutoRCCar #157: Ultrasonic sensor problem

sundar19, 2019-03-03:
Mr.hamuchiwa
Even if the Ultrasonic sensor is detected the car doesn't stop.pls help

sundar19, 2019-03-04:
Mr. Hamuchiwa I also have another problem, I included Cascade classifiers for left arrow and detected it,but the problem is I can't steer the vehicle in accordance as def(self,prediction) does not have a special condition for taking left . This is in rc_driver_helper.py file. Please tell me how to add our own steer() condition
Thank you

sundar19, 2019-03-04:
Mr.hamuchiwa,
After i run collect_training_data.py and collect data and i convert .npz file to .xml file and run rc_driver.py the car doesn't follow the lane and goes in a wrong direction. i want to know,
1. what will be the height of the camera from the car?
2.what should be the color of the lane and the road?
Thank You

hamuchiwa, 2019-03-04:
Steering function call is in `rc_driver.py`, feel free to add your functions to handle special steering case. 
Color doesn't matter, image is converted to grayscale before saving. Check out #139 if you have doubts on training.

closed 2019-03-04 by hamuchiwa

kxrf97, 2019-03-06:
@sundar19 my car keep running randomly after training..... did you get the car going well?

sundar19, 2019-03-08:
If you want to train your car for the lane
1. you should speed control your car
2. you should good streaming speed
3.try adding more hidden layers to your neural network
4. if all these don't work you need to go for convolutional neural network using tensorflow

kxrf97, 2019-03-10:
@sundar19 how do i add more hidden layers to my neural network? i am new to this.....no really know how to do in the coding part

sundar19, 2019-03-10:
https://docs.opencv.org/2.4/modules/ml/doc/neural_networks.html#cvann-mlp-trainparams
This will help you.