hamuchiwa/AutoRCCar #160: Improvisation

akilystic, 2019-03-05:
Is there a way by which I can stream the serial commands back to the raspi on the car and later read it via Arduino connected to the raspi? I want to eliminate the rc part. Please help me with this.

sundar19, 2019-03-07:
https://github.com/ryangmolina/rpi-car-arrow-detection-using-cascade-classifier?files=1

akilystic, 2019-03-07:
hey sundar.. thanks a lot. so this is fully based on the pi right? No local computer needed? Any pre requisites/ tips before implementing this on the pi? just run it straightway on the pi?

sundar19, 2019-03-07:
No you should have a PC and live stream video of camera connected to rpi will be viewed in PC and USB of pc will control remote

akilystic, 2019-03-07:
So what's the difference in the link you've posted?

sundar19, 2019-03-08:
That link will give you an idea of directly connecting motors with pi and controlling .