hamuchiwa/AutoRCCar #162: TURN RIGHT

hobnob453, 2019-03-05:
I have followed the instructions as part of a college project, after managing to set up all of the python code, i have trained the car with 6 .npz files which modify the nm_model.xml, however when i run the main program (rc_driver) along with stream and ultrasonic files on the PI, the car just wants to turn right all of the time, i will continue to add training data but could do with some guidance as to where i could be going wrong? Thanks,

kxrf97, 2019-03-06:
i also face the same problem.......

kxrf97, 2019-03-06:
did you manage to tackle it?>

hobnob453, 2019-03-06:
Having another go tonight, just gonna train it some more, glad its not just me doing something stupid, how much ‘collect_training’ have you done?

sundar19, 2019-03-07:
I think something wrong with your wheel alignment maybe and check in the Python shell for forward commands and inspite of forward commands sent your car turns right then it's wheel alignment problem!

kxrf97, 2019-03-07:
I collect around 300 images, and my accuracy is around 60+ after training.... But my car just keep going randomly

kxrf97, 2019-03-07:
@sundar19 
Wheel alignment are having no problem, the command prompt keep showing right , should not be the alignment problem. Did you manage to finish this project.

hobnob453, 2019-03-07:
Yes my shell is sending right commands, only right commands

sundar19, 2019-03-08:
@kxrf97  I used Cascade classifier of left direction sign to finish my project . The neural network you use will be efficient only when the streaming rate is good. Switch off the car and move the car by hand whether the camera lags,if that's the case Cascade classifier is the only solution

kxrf97, 2019-03-08:
Can you send me any tutorial about that? How you use cascade classifier to finish the project. My streaming rate is abit laggy... I guess no smooth as i think...

kxrf97, 2019-03-08:
Normally cascade classifier is use for object detection. How can you apply om the track

hobnob453, 2019-03-08:
Yes i would love to know how you changed the classifier, again its not the smoothest but the connection should be fast enough.

sundar19, 2019-03-08:
https://github.com/ryangmolina/rpi-car-arrow-detection-using-cascade-classifier?files=1

Take Cascade classifier for left or right sign from this link and include your right or left turn functions in rc_driver_helper.py

somalwar, 2019-03-17:
@hamuchiwa I am also facing the same issue. Every prediction is a 'RIGHT' turn. The training and validation accuracy reaches around 90% with about a hundred images, yet it always makes a 'RIGHT' prediction. Any suggestions? Thanks!

somalwar, 2019-03-17:
Yes I am facing the same issue, it always wants to make a 'right' turn. However on a different training set, it always made a 'forward' prediction. And on yet another one, where I trained it to only go 'forward' or 'left', it did output 'forward' and 'left' but the predictions were not correct at all. Not sure if its a training problem or something else.

VishalWrench, 2019-03-29:
@somalwar have u solved that problem

somalwar, 2019-03-29:
@VishalWrench No I haven't. I plan to use Tensorflow in the future, maybe that will give better results.

VishalWrench, 2019-03-29:
@somalwar thanks for replying yeah i am also trying to implement this in keras

jhadkaterji, 2019-04-01:
@somalwar @VishalWrench Did either of you guys end up figuring out how to get the car to stop just going right all the time? was it an issue with the training?

jhadkaterji, 2019-04-01:
> @somalwar @VishalWrench Did either of you guys end up figuring out how to get the car to stop just going right all the time? was it an issue with the training?

Im having the same issue and cant figure out whats wrong