hamuchiwa/AutoRCCar #164: RC control test

SaiSurya9999, 2019-03-13:
I am unable to find the point to which i have to solder my audrino pins could you please help me to find to what points i should connect my forward backward left right in remote...??
[image]
[image]

SaiSurya9999, 2019-03-13:
Please respond ASAP I have uploaded HD photos of my transmitter chip just need to know which points should we connect or solder for forward, backward, left , right functions..??
************Thank you*************

VishalWrench, 2019-03-13:
### I have found my rc car tactile switches so i sorted out which pins of switch is connected to the board  and used optocouplers (4N35) to couple different circuits.I have provided schematic of one switch just replicate same for remaining switches .Hope this will  help u
[image]
[image]
[image]
[image]

VishalWrench, 2019-03-13:
### This is my arduino sketch ignore the comments (//)(italic in the code)
// assign pin num
int right_pin = 6;
int left_pin = 7;
int forward_pin = 10;
int reverse_pin = 9;
/*int f1 = A2;
int re1 = A3;
int l1 = A4;
int r1 = A5;*/

// duration for output
int time = 50;
// initial command
int command = 0;

void setup() {
  pinMode(right_pin, OUTPUT);
  pinMode(left_pin, OUTPUT);
  pinMode(forward_pin, OUTPUT);
  pinMode(reverse_pin, OUTPUT);
 /* pinMode(f1, OUTPUT);
  pinMode(re1, OUTPUT);
  pinMode(l1, OUTPUT);
  pinMode(r1, OUTPUT);*/
  digitalWrite(forward_pin, LOW);
  digitalWrite(reverse_pin, LOW);
  digitalWrite(right_pin, LOW);
  digitalWrite(left_pin, LOW);
  /*digitalWrite(f1, LOW);
  digitalWrite(re1, LOW);
  digitalWrite(r1, LOW);
  digitalWrite(l1, LOW);*/
  Serial.begin(115200);
}

void loop() {
  //receive command
  if (Serial.available() > 0){
    command = Serial.read();
  }
  else{
    reset();
  }
   send_command(command,time);
}

void right(int time){
  digitalWrite(right_pin, HIGH);
  //digitalWrite(r1, HIGH);
  delay(time);
 // digitalWrite(right_pin, LOW);
  //digitalWrite(r1, LOW);
  
}

void left(int time){
  digitalWrite(left_pin, HIGH);
 // digitalWrite(l1, HIGH);
  delay(time);
 // digitalWrite(left_pin, LOW);
  //digitalWrite(l1, LOW);
}

void forward(int time){
  digitalWrite(forward_pin, HIGH);
  //digitalWrite(f1, HIGH);
  delay(time);
  //digitalWrite(forward_pin, LOW);
  //digitalWrite(f1, LOW);
}

void reverse(int time){
  digitalWrite(reverse_pin, HIGH);
  //digitalWrite(re1, HIGH);
  delay(time);
  //digitalWrite(reverse_pin, LOW);
  //digitalWrite(re1, LOW);
}

void forward_right(int time){
  digitalWrite(forward_pin, HIGH);
  //digitalWrite(f1, HIGH);
  digitalWrite(right_pin, HIGH);
  //digitalWrite(r1, HIGH);
  delay(time);
 // digitalWrite(forward_pin, LOW);
  //digitalWrite(f1, LOW);
  //digitalWrite(right_pin, LOW);
  //digitalWrite(r1, LOW);
}

void reverse_right(int time){
  digitalWrite(reverse_pin, HIGH);
  //digitalWrite(re1, HIGH);
  digitalWrite(right_pin, HIGH);
  //digitalWrite(l1, HIGH);
  delay(time);
  //digitalWrite(reverse_pin, LOW);
  //digitalWrite(re1, LOW);
  //digitalWrite(right_pin, LOW);
  //digitalWrite(l1, LOW);
  
}

void forward_left(int time){
  digitalWrite(forward_pin, HIGH);
  //digitalWrite(f1, HIGH);
  digitalWrite(left_pin, HIGH);
 // digitalWrite(l1, HIGH);
  delay(time);
  //digitalWrite(forward_pin, LOW);
  //digitalWrite(f1, LOW);
  //digitalWrite(left_pin, LOW);
  //digitalWrite(l1, LOW);
}

void reverse_left(int time){
  digitalWrite(reverse_pin, HIGH);
  //digitalWrite(re1, HIGH);
  digitalWrite(left_pin, HIGH);
  //digitalWrite(l1, HIGH);
  delay(time);
  //digitalWrite(reverse_pin, LOW);
  //digitalWrite(re1, LOW);
  //digitalWrite(left_pin, LOW);
  //digitalWrite(l1, LOW);
}                                                                                                                                                                                        

void reset(){
  digitalWrite(right_pin, LOW);
  digitalWrite(left_pin, LOW);
  digitalWrite(forward_pin, LOW);
  digitalWrite(reverse_pin, LOW);
  /*digitalWrite(l1, LOW);
  digitalWrite(r1, LOW);
  digitalWrite(re1, LOW);
  digitalWrite(f1, LOW);*/
}

void send_command(int command, int time){
  switch (command){

     //reset command
     case 0: reset(); break;

     // single command
     case 1: forward(time); break;
     case 2: reverse(time); break;
     case 3: right(time); break;
     case 4: left(time); break;

     //combination command
     case 6: forward_right(time); break;
     case 7: forward_left(time); break;
     case 8: reverse_right(time); break;
     case 9: reverse_left(time); break;

     default: Serial.print("Inalid Command\n");
    }
}

MyFirstRoboticProject, 2019-03-13:
I solved this issue by removing the buttons and soldering the wires on to the circular middle part. Keep it in mind that the circular middle part and the surrounding area shouldn't be in contact and should be electrically isolated.

MyFirstRoboticProject, 2019-03-13:
There should be a single tape applied on the buttons. Find it on the board and remove it. Be careful though

SaiSurya9999, 2019-03-14:
Did you keep batteries in the remote or audrino is sending power through pins...???

MyFirstRoboticProject, 2019-03-14:
I powered it from 3.3V arduino

SaiSurya9999, 2019-03-17:
@MyFirstRoboticProject  Could you provide pics of the RC car and your remote circuit so that i can understand clearly please...!!!

SaiSurya9999, 2019-03-18:
> ### I have found my rc car remote controller has tactile switches so i sorted out which pins of switch is connected to the board and used optocouplers (4N35) to couple different circuits i.e. arduino and controller .I have provided schematic of one switch just replicate same for remaining switches .Hope this will help u
> [image]
> [image]
> [image]
> [image]

Very much helpful Thank you so much...!!

SaiSurya9999, 2019-03-19:
> ### This is my arduino sketch ignore the comments (//)(italic in the code)
> // assign pin num
> int right_pin = 6;
> int left_pin = 7;
> int forward_pin = 10;
> int reverse_pin = 9;
> /_int f1 = A2; int re1 = [...]

Could you please provide me the pics of audrino and breadboard clearly for one function forward so that i could do better..

VishalWrench, 2019-03-19:
### this is schematic of my circuit. just ignore those comments and extra pins (f1,r1 ....) connect the circuit and burn the arduino sketch into uno and run rc_control_test.py. It will work
[image]

SaiSurya9999, 2019-03-19:
Thank you so much i have sorted out everything with your help..!!!
Mine is working fine.... 🥇

Sirajabudabe, 2019-03-28:
Hi @SaiSurya9999 
could you please send me your collected data file ?
i tried every thing and i can't get the accuracy of neural network over 60% , did you face this problem to if so how did you solve it?
thank you in advance

nandhukrishnan, 2019-04-02:
> Thank you so much i have sorted out everything with your help..!!!
> Mine is working fine....

### **Hi @SaiSurya9999 I have exactly same remote control as yours. Could you please send me your connection diagram?  Thanks in advance.**

nandhukrishnan, 2019-04-03:
> I am unable to find the point to which i have to solder my audrino pins could you please help me to find to what points i should connect my forward backward left right in remote...??
> [image]
> ![WhatsApp Image 2019-03-13 at 8 30 17 PM 

Please help me to make a connection diagram for this board

arunph, 2019-04-10:
Hi @saisurya9999 plz help me out with the connection pic of ur rc.since I m having the same RC control

nandhukrishnan, 2019-04-12:
@SaiSurya9999 Can you provide your mail id? I need help in the connection diagram of remote  control. Its urgent...!!!

arunph, 2019-04-12:
pharun0059@gmail.com

arunph, 2019-04-12:
Tqs

nandhukrishnan, 2019-04-20:
> Hi @SaiSurya9999 plz help me out with the connection pic of ur rc.since I m having the same RC control

Hi..Did you got this?

arunph, 2019-04-21:
No I didn't got

codeskul, 2019-04-25:
I have used octocupler for push button switch in my project but issue is it take only reverse and left please help me it's urgent

arunph, 2019-04-25:
Can u send the clear pic of ur circuit connection

codeskul, 2019-04-25:
[image]
[image]
[image]

arunph, 2019-04-25:
send me the  whole pic of your connection where you had used 4 optocouplers

<https://www.avast.com/en-in/recommend?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=default3&tag=56ab1e3c-1ede-4e42-b16f-4564b64aa948>
I’m
protected online with Avast Free Antivirus. Get it here — it’s free forever.
<https://www.avast.com/en-in/recommend?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=default3&tag=56ab1e3c-1ede-4e42-b16f-4564b64aa948>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

codeskul, 2019-04-25:
I will send you whole image of my circuit, i have one question that it require any registers between octocupler and push button of remote ?

arunph, 2019-04-25:
No it's ok to use without resistors

codeskul, 2019-04-25:
[image]
[image]

codeskul, 2019-04-25:
It done successfully

arunph, 2019-04-25:
Okay what was the error

arunph, 2019-04-25:
> > Hi @SaiSurya9999 plz help me out with the connection pic of ur rc.since I m having the same RC control
> 
> Hi..Did you got this?

you can refer the code skull posted images for final working circuit diagram

nandhukrishnan, 2019-04-25:
> > > Hi @SaiSurya9999 plz help me out with the connection pic of ur rc.since I m having the same RC control
> > 
> > 
> > Hi..Did you got this?
> 
> you can refer the code skull posted images for final working circuit diagram

> I am unable to find the point to which i have to solder my audrino pins could you please help me to find to what points i should connect my forward backward left right in remote...??
> [image]
> [image]

**But I didn't understand which pins to solder on this board? This board is exactly same as mine
. Plz help me** @arunph @codeskul @SaiSurya9999

nandhukrishnan, 2019-04-26:
> I solved this issue by removing the buttons and soldering the wires on to the circular middle part. Keep it in mind that the circular middle part and the surrounding area shouldn't be in contact and should be electrically isolated. There should be a tape applied on the buttons. Find it on the board and you have to remove it to take the buttons off the board. Good luck!

Can you provide a pic of your connection?

nandhukrishnan, 2019-04-28:
> **But I didn't understand which pins to solder on this board? This board is exactly same as mine . Plz help me** @arunph @codeskul @SaiSurya9999

[image]
**Yeahhh..!!! I got it.**

arunph, 2019-04-28:
Well done.

arunph, 2019-04-29:
> [image]
> **Yeahhh..!!! I got it.**

Is the car running using your keyboard.and how did you made it.can u send the circuit connection pic of it

arunph, 2019-04-29:
> It done successfully thank you very much

hey its urgent can you help me out.when i connect wire to the aurdino GND car start moving automatically without pressing of arrow keys in the keyboard.

nandhukrishnan, 2019-04-30:
> > It done successfully thank you very much
> 
> hey its urgent can you help me out.when i connect wire to the aurdino GND car start moving automatically without pressing of arrow keys in the keyboard.

@arunph I also had the same problem. But it was due to the arduino program. You just run the code above by @VishalWrench . Mine is working fine now 😊

nandhukrishnan, 2019-04-30:
> > It done successfully thank you very much
> 
> hey its urgent can you help me out.when i connect wire to the aurdino GND car start moving automatically without pressing of arrow keys in the keyboard.

> Is the car running using your keyboard.and how did you made it.can u send the circuit connection pic of it

[image]
[image]

arunph, 2019-05-01:
Thank you very much.but is it working

arunph, 2019-05-04:
First it was working fine.but now again it is getting the same thing like
car start moving automatically without pressing of keyboad keys

arunph, 2019-05-06:
hey kindly help me with.excution steps of program.how to train and test.

<https://www.avast.com/en-in/recommend?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=default3&tag=56ab1e3c-1ede-4e42-b16f-4564b64aa948>
I’m
protected online with Avast Free Antivirus. Get it here — it’s free forever.
<https://www.avast.com/en-in/recommend?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=default3&tag=56ab1e3c-1ede-4e42-b16f-4564b64aa948>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

<https://www.avast.com/en-in/recommend?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=default3&tag=56ab1e3c-1ede-4e42-b16f-4564b64aa948>
I’m
protected online with Avast Free Antivirus. Get it here — it’s free forever.
<https://www.avast.com/en-in/recommend?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=default3&tag=56ab1e3c-1ede-4e42-b16f-4564b64aa948>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

arunph, 2019-05-07:
hey anyone finished this project.plz help me out with steps

145shahzad, 2019-06-01:
[image]
i'm having this error when i run rc_control_test....i have also coneccted arduino with pc and car's remote.....can anyone help??

arunph, 2019-06-02:
Kindly change 'dev/tty.usbmodem' with ur arduino port name like com5,etc.just once go through ur arduino IDE .Then go to tools,there u will see port option.just check on that option.which port ur ardunio is using like com3,com4,etc.then that same port name should be mentioned in that line by removing this dev/try.usbmodem1421.

145shahzad, 2019-06-02:
thanks, program unned with zero error.....but i have remote control car in
which push buttons are used......i have conneted wires from breadboard to
arduino.....when i press keys the car is not moving
controller light blinks but car shows no response

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

arunph, 2019-06-03:
Can u send  snapshot of ur  connection from breadboard to arduino

harshithdayananda, 2019-06-06:
@nandhukrishnan i need your help. can you mail me at harshith.daya@gmail.com?

145shahzad, 2019-06-06:
I have a issue......mine is the same controller as @SaiSurya9999 .....but when i solder all the wires it only moves in Forward left , forward right, reverse left and reverse right.......it shows no actions on forward, reverse, left and right.......i have used 4N35 and arduino mega

145shahzad, 2019-06-07:
Hi.....can you please tell me how you connect ultrasonic sensor to arduino
or Raspberry pi?
Please tell me about connection of ultrasonic sensor..

arunph, 2019-06-08:
> I have a issue......mine is the same controller as @SaiSurya9999 .....but when i solder all the wires it only moves in Forward left , forward right, reverse left and reverse right.......it shows no actions on forward, reverse, left and right.......i have used 4N35 and arduino mega

i think its better to use arduino UNO.then it will work fine

145shahzad, 2019-06-09:
Arduino mega is not best for this?

arunph, 2019-06-10:
Yaa.but once check ur arduino code and connection

BonthuAI, 2019-08-30:
Hi All
can you please let me know
1. whether you disconnected the batteries of remote?, I have RC remote with 12V battery - so need some clarification
2. has anyone done this similar one using raspberry PI ?

arunph, 2019-09-03:
no,don't remove the battery of remote

BonthuAI, 2019-09-03:
Thanks Arun,

however, The images posted by code Skull doesn't have batteries connected to RC circuit board
Am I missing anything ? 

(Typo in my earlier post, read it is 9V instead of 12V)

Also I am looking for some inputs for the same task for Raspberry instead of Arduino 

Regards
Srinivas B

BonthuAI, 2019-09-05:
thanks to the participants of this thread
with the help of this thread, we successfully completed the similar task using Raspberry pi 4

maazmmd, 2020-01-31:
> @arunph I also had the same problem. But it was due to the arduino program. You just run the code above by @VishalWrench . Mine is working fine now 😊
>
> [image]
> [image]

Hey @nandhukrishnan could you please tell me what is the black IC that is connected? as I was unable to co-relate that