hamuchiwa/AutoRCCar #166: Rc_driver.py issue

sundar19, 2019-03-14:
Mr.hamuchiwa ,
rc_control_test is working fine but when I run rc_driver.py no forward is displaying on the shell and commands are not sent pls help.

hamuchiwa, 2019-03-15:
@sundar19 I just uploaded another file called `rc_driver_nn_only.py` which eliminates object detection portion. Can you try it out?

somalwar, 2019-03-15:
hi @hamuchiwa , is the rc_driver_nn_only.py file just used for troubleshooting? Also, I see you made some edits to rc_driver.py file. Could you please comment on that. Thanks!

hamuchiwa, 2019-03-15:
@somalwar There are people asking for a short version, and mostly object detection is not an interest anyway. So it's for running on track only, and yes it can be used for troubleshooting as well. Regarding rc_driver edit which is really a tiny change, used variables to replace some repetitive numbers.

sundar19, 2019-03-16:
Thanks Mr.hamuchiwa mine is working fine now!

sundar19, 2019-03-16:
And how to fix the distance for object detection my camera sometimes detects stop sign when it's in a very long distance or sometimes in a very short distance . Which parameter or variable should I change to fix my distance from which object should be detected? @hamuchiwa

hamuchiwa, 2019-03-18:
@sundar19 You can change the threshold at line 50 in `rc_driver.py`.

sundar19, 2019-03-18:
Thanks @hamuchiwa

SaiSurya9999, 2019-03-19:
> And how to fix the distance for object detection my camera sometimes detects stop sign when it's in a very long distance or sometimes in a very short distance . Which parameter or variable should I change to fix my distance from which object should be detected? @hamuchiwa

Could you tell me where did you get the stop sign and if possible even traffic light...Any links..??

closed 2019-03-27 by hamuchiwa

ShwetaLalwani, 2019-05-09:
@sundar19 Can you plz share your trained . XML file ? It's really urgent.

sundar19, 2019-05-10:
@ShwetaLalwani  my xml is of no use to you. My neural network data doesn't work fine. So I used only forward key to train my neural network and used haar Cascades of left and right  to turn my car. So it is better that you use tensorflow with CNN and more hidden layers and train your xml.