hamuchiwa/AutoRCCar #168: How do you have training techniques?

PuriHa, 2019-03-17:
How do you have training techniques? 
I should press forward, left, right and hold or press it several times?
When I want the rc car to turn right, How should I press?
I try to do everything possible. But my car still doesn't turn correctly!
[image]
[image]

Please tell me, I love this project very much, and I am very thoughtful. Can you answer my question? Thank you very much.

hamuchiwa, 2019-03-18:
@PuriHa Besides the actual model tuning, keep your car centered, make sure images are mostly labeled correctly and your data is mostly balanced. If there are too many "forward" labels, try to drive on corner sections more. In the case the car is not centered, collecting some error correction data can also help.

You can try to tune the model in OpenCV as well, but I think what's better and simpler now is to create a CNN model in Tensorflow/Keras.

PuriHa, 2019-03-23:
If I use Tensorflow, what steps should I do? How to do? 
Please help explain to me thoroughly. And then Code?
Thanks very much for your advice @hamuchiwa

VishalWrench, 2019-03-23:
https://www.youtube.com/playlist?list=PLQVvvaa0QuDfhTox0AjmQ6tvTgMBZBEXN
may be this deep learning tutorial (CNN) using tensor-flow and Keras by Sentdex will help you. You can find many resources on internet about CNN. There is a whole series of self driving car (python plays gta 5 ) on his channel check it out.

PuriHa, 2019-04-04:
And then file code **model.py** ? What can do? In what step?