hamuchiwa/AutoRCCar #172: Pi camera calibiration

SaiSurya9999, 2019-03-21:
My Camera is Raspberry pi 3 v2 camera module officially distributed by PI manufacturers..
Should i need to do the calibiration if yes please tell me how to do that i am not understanding...
I have attached my car pics below please observe the camera in that..
[image]
[image]

**Thank you @hamuchiwa ..!!******

SaiSurya9999, 2019-03-21:
And one more doubt can i train this in a circular paper track (Exact track as you shown in the video borders with paper but circular)  if yes then how many test images it requires i mean you told 300 images are sufficient for your track approx how many for circular track...!!!
I know we have to balance the model without biased labels like more forward more right or left..
Thank you @hamuchiwa ..!!

hamuchiwa, 2019-03-21:
@SaiSurya9999 Your car looks very nice! I had pi camera v1 I believe, so you may need calibration, but the distance measured using a single camera isn't not really accurate, for approximation it's fine. 

I think for a circular track, it's better to drive around the track for two or three times.
Here is a video made by "aha124226675" https://www.youtube.com/watch?v=KrvvmLEQk8o

SaiSurya9999, 2019-03-24:
> @SaiSurya9999 Your car looks very nice! I had pi camera v1 I believe, so you may need calibration, but the distance measured using a single camera isn't not really accurate, for approximation it's [...]

Thank you @hamuchiwa  please respond to issue no #174

closed 2019-03-27 by hamuchiwa