hamuchiwa/AutoRCCar #173: TRAINING...

SaiSurya9999, 2019-03-21:
[image]
-- I have few doubts in training all set hardware is assembled and at last came to the training phase.
-- I have uploaded an image where you can see my training track.
-- Should we do the total training session in one execution of collect_training.py..??
-- For each execution there is an npz file created if we use one npz file the previous npz file which we trained is dumped..??
-- Could you give me your 300 images npz file or collection of those so that i can use them and train my model fast with more data.
-- When i place my npz file in model_test pynb file then i get some size error i have uploaded that below.
[image]

SaiSurya9999, 2019-03-21:
For stopping condition is there a break like reverse signal once and then idle i dont see any such condition..??
class RCControl(object):

    def __init__(self, serial_port):
        self.serial_port = serial.Serial(serial_port, 115200, timeout=1)

    def steer(self, prediction):
        if prediction == 2:
            self.serial_port.write(chr(1).encode())
            print("Forward")
        elif prediction == 0:
            self.serial_port.write(chr(7).encode())
            print("Left")
        elif prediction == 1:
            self.serial_port.write(chr(6).encode())
            print("Right")
        else:
            self.stop()

    def stop(self):
        self.serial_port.write(chr(0).encode())
-->what lines could we add to stop call to break the speed by reverse signal and then execution line **break;**         ..

hamuchiwa, 2019-03-22:
@SaiSurya9999 Make sure to slow down the car before collecting data, if it runs too fast, the video streaming couldn't keep up. 
You can collect multiple times, in training it will loads all files in `training_data` folder. 
This is an old project, I don't even have the car anymore. 
 Check the input size, in notebook it's 240*320, your image may be 120*320.

SaiSurya9999, 2019-03-24:
[image]
How to increase the validation accuracy..??
I have uploaded the image of my track where i have drove it from both sides many times approx 350 frames were recorded..But there is no increase in validation accuracy of training..
It is biased to forward please help me @hamuchiwa 
[image]

SaiSurya9999, 2019-03-24:
I have slowed the vehicle so no problem about speed or stream rate cope up..

hamuchiwa, 2019-03-25:
@SaiSurya9999 You can collect multiple times and then train together. If you are satisfied what you have collected, yes, then you need to delete that file before training again.

Sirajabudabe, 2019-03-26:
Hi 
I have the same problem i can't increase the accuracy of my training i tried a lot but i can't get above 70% 
could you please upload your .npz so i can try it.
Thanx

hamuchiwa, 2019-03-27:
@Sirajabudabe Since it's duplicated with another thread of yours, I'll close the other one. Can you upload your data?

SaiSurya9999, 2019-03-27:
How to reset the neural network model as a baby to train it again with nice balanced data..

hamuchiwa, 2019-03-27:
@SaiSurya9999 All model related stuff is in `model.py`, you can create your own architecture there. Every time you run `model_training.py`, it will overwrite the old one with the same file name `nn_model.xml`.

SaiSurya9999, 2019-03-28:
What is your training accuracy when you successfully made the car play with corners..??
(I mean car taking successful turns at right point.)

Sirajabudabe, 2019-03-28:
@hamuchiwa  https://drive.google.com/drive/folders/1D2WbSvCy1LnNEoIAA69qk_IvJFVrXvgO

hamuchiwa, 2019-03-29:
@Sirajabudabe Can you remove the permission, I'll let you know after downloading, then you can delete the file.

Sirajabudabe, 2019-03-31:
https://drive.google.com/file/d/1kcK2YEr8PdTN50MhcPSfWqxI0GHSEC-a/view?usp=sharing
could you please run it on your neural network and tell me what you got

hamuchiwa, 2019-04-01:
@Sirajabudabe I have uploaded a jupyter notebook using keras. After a quick run, as you can see from the plots, the model is overfitting, I'll leave it to you for tuning. Good luck!
[Untitled.ipynb.zip](https://github.com/hamuchiwa/AutoRCCar/files/3030069/Untitled.ipynb.zip)

Sirajabudabe, 2019-04-02:
@hamuchiwa  sorry but am not really familiar to deep learning and machine learning techniques , how can i run this model 
is the problem in cv2 neural network ?
sorry again for my many questions

PuriHa, 2019-04-02:
Then file Untitled.ipynb.zip, how does it apply to training?

hamuchiwa, 2019-04-02:
@Sirajabudabe @PuriHa It's in jupyter notebook format, unzip it, then open it in jupyter. To open it, start a command line window, then do `jupyter notebook`. Once the browser is open, navigate to the file. For more details about jupyter, you can refer to the doc or search some tutorials.

PuriHa, 2019-04-03:
@hamuchiwa And what should I do with the file Untitled.ipynb,zip? I should try to run the code in the file? I don't know, how does it apply to training? How can I run the code in a file jupyter notebook? And what are the steps? Why didn't the code be put in the Github? Provide clear codes and procedures.
thank you for answers @hamuchiwa

rahulkvr, 2019-04-09:
@hamuchiwa the Untitled.ipynb.zip gives trained model as model_test.h5. 
In the rc_driver_nn_only.ipynb the trained model named nn_model.xml is used. What are the changes required in rc_driver_nn_only.ipynb for taking model_test.h5. file.

Sirajabudabe, 2019-04-10:
@hamuchiwa  My question is the same as @rahulkvr  how to use the Untiteld.ipynb file in rc_driver_nn_only ???

hamuchiwa, 2019-04-10:
@Sirajabudabe I think I have included the code on how to load model and make predictions in the notebook. Since the model is in Keras not in OpenCV, so you need to modify the code of loading and predicting in  rc_driver_nn_only.py.

rahulkvr, 2019-04-10:
@hamuchiwa  Sir can you please share the modified jupyter notebook file with code for loading and predicting using Keras.

hamuchiwa, 2019-04-10:
@rahulkvr It's in the notebook.

rahulkvr, 2019-04-11:
@hamuchiwa  I want to run the car using the saved model of keras. But in the rc_driver_nn_only.py it is not possible. Can you modify that particular python code and upload here.

Sirajabudabe, 2019-04-21:
@SaiSurya9999  could you please upload your training model, or the parameters of neural network (hidden layers iterations ....etc) thanks in advance @hamuchiwa

Razor456, 2019-05-14:
hello
we need help guys @hamuchiwa 
Our rc car is just turning around after doing the training ! we got 59% accuracy after the training !

One more porblem we've had is our rc car speed. how did you slow down your rc car's speed ?

Best regards

Sirajabudabe, 2019-09-05:
Dear @SaiSurya9999  could you please contact me on e-mail : serajabudaber@gmail.com

9salman7, 2020-03-02:
@rahulkvr did you manage to solve the issue?

rahulkvr, 2020-03-03:
> @rahulkvr did you manage to solve the issue?

No, I couldn't resolve it.