hamuchiwa/AutoRCCar #174: Obstacle detection..!!

SaiSurya9999, 2019-03-21:
[image]
When i run RC_DRIVER.py by placing an obstacle it is detecting but not stoping the car and  predictions of turning did not stop as you see in the screen shot.!!
-- A small doubt how to reduce the speed of RC car as my track is small car is responding too fast which passes the track in second more prone to error... If i increase time variable value from 50 to 250 i mean instruction delay will it solve my problem..??
Please respond ASAP ....
Thank you @hamuchiwa

SaiSurya9999, 2019-03-24:
Please respond to this @hamuchiwa ...

hamuchiwa, 2019-03-25:
@SaiSurya9999 The console output suggests the car stopped only for one frame, then distant is not being sent until next measurement. Try set time interval to a lower number, eg. 0.1 in `ultrasonic_client.py` line 53.

SaiSurya9999, 2019-03-27:
Thank you so much @hamuchiwa..!!

hamuchiwa, 2019-03-27:
@SaiSurya9999 Can you post the error?

SaiSurya9999, 2019-03-27:
@hamuchiwa i finally found a solution.. Thank you for the assistance without your suggestion it may not be that easy to find it..
-- I have used 0.1 in place of 0.5 in ultra_client.py but it gave me socket error as host reset the connection.
-- So i used another variable called 'alert' and stored the sensor data which is being reset for every 0.5 sec that helped me to solve delay of data by using previous ultrasonic sensor data.
Thank you @hamuchiwa

SaiSurya9999, 2019-03-28:
This is where i changed code...
from model import NeuralNetwork
from auto_helper import *

# distance data measured by ultrasonic sensor
sensor_data = None
alert = 100

class SensorDataHandler(socketserver.BaseRequestHandler):

    data = " "

    def handle(self):
        global sensor_data
        global alert

        while self.data:
            self.data = self.request.recv(1024)
            sensor_data = round(float(self.data), 1)
            alert = sensor_data
            # print "{} sent:".format(self.client_address[0])
            print(alert)
//In obstacle check
if alert < 30:
                        print("Stop, obstacle in front")
                        self.rc_car.stop()
                        sensor_data = None

SaiSurya9999, 2019-03-28:
@hamuchiwa could you please tell me how to execute two python scripts(I mean stream.py and ultra.py in pi terminal). It may be silly question but i tried googling it and found solutions like shell scripts but dont have any knowledge about them could you please help me out..
Thank you @hamuchiwa

hamuchiwa, 2019-03-28:
@SaiSurya9999 You can open two terminals on pi.

closed 2019-04-12 by hamuchiwa