hamuchiwa/AutoRCCar #175: Question about stream rate

lz130dd, 2019-03-22:
Mr. Hamuchiwa,
For the collecting_training_data part, I didn't find your sample rate of key input, and usually the stream rate of key input is larger than the stream rate of image.
So how do you make sure every frame of image can has a key input label?

hamuchiwa, 2019-03-22:
Yes, image is dropped if there is no key input.

closed 2019-03-27 by hamuchiwa

lz130dd, 2019-04-16:
您好， 抱歉再次打扰您，我在训练神经网络的时候遇到一些问题想请教您一下：
在600~1000frames 的准确率一直在50%~70%之间，无法准确预测。
我是用ROS系统在一个扫地机器人上搭的平台。我的机器人直走时一直有点右偏，所以采数据的时候直线一直有按左键修正，不知道跟这个有没有关系。
还有就是我采数据时车速较慢，采完一个直线加左右转弯就差不多有近300frames了（摄像头帧数在10HZ出头，延迟状况还可以），是不是数据量不太够？应该加快一点车速多采几组吗？（有点担心采多了模型过载）
如果还是不行我准备试试您在jupyter note 上的keras模型， 用cnn会不会准确率会高一些？您的keras模型我可以直接拿来用吗？（时间紧迫）

hamuchiwa, 2019-04-16:
@lz130dd 时间紧迫不如直接上CNN吧，那个notebook是基于一个小哥的数据，凭着感觉加了几层，到时训练的时候你可以再做调整。如果跑道不换的话，模型过拟合没关系，训练的数据也可以不用分成train和valid，这样训练的数据还能多一点儿。
总之多跑几次吧，修正偏离的情况也可以来几次，慢慢的跑就行。

lz130dd, 2019-04-16:
@hamuchiwa 我尝试了一下用keras 不知道是不是参数问题 为什么200多frames 很快就训练完了，您当时用keras会比用opencv快很多吗

hamuchiwa, 2019-04-16:
@lz130dd 是快的，多线程吧，有GPU那就更快了。

lz130dd, 2019-04-17:
@hamuchiwa 还想请问一下您，您当时用opencv准确率大概有多少，是固定用一条左右前比较平均的轨道采数据吗？您训练完成后可以在新的未训练的轨道上自动驾驶吗？

hamuchiwa, 2019-04-17:
@lz130dd 具体的数记不太清了，0.85左右吧？右转没有收集，直行左转摆了好多次，最后摆的新轨道录得视频。不过我没换地点，都在同一条走廊上，再就是在车跑出末端之前我就不收集了。
有个在环形轨道上驾驶的视频不知道你看过没，他那个跑的就挺好的，