hamuchiwa/AutoRCCar #176: issue after i run rc_driver.py

Sirajabudabe, 2019-03-22:
I have a broken pipe error every time i run Rc_driver.py , i didn't have this problem when i run training and testing codes , all tho the part of code responsible of connection  is the same

hamuchiwa, 2019-03-22:
Can you post the error on your computer as well as on raspberry pi? 
Have you tried to run `rc_driver_nn_only.py`? Does it give you error as well?

Sirajabudabe, 2019-03-24:
The rc_driver_nn_only.py work just fine after i change the host to " self.server_socket.bind(("0.0.0.0", port))"
in the other hand the rc_driver.py don't work as well , i got this error 
Traceback (most recent call last):
  File "rc_driver.py", line 188, in <module>
    ts.start()
  File "rc_driver.py", line 181, in start
    self.video_stream(self.host, self.port1)
  File "rc_driver.py", line 170, in video_stream
    s = socketserver.TCPServer((host, port), VideoStreamHandler)
  File "/home/siraj/miniconda3/envs/auto-rccar/lib/python3.7/socketserver.py", line 452, in __init__
    self.server_bind()
  File "/home/siraj/miniconda3/envs/auto-rccar/lib/python3.7/socketserver.py", line 466, in server_bind
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/home/siraj/miniconda3/envs/auto-rccar/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/home/siraj/miniconda3/envs/auto-rccar/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "rc_driver.py", line 174, in sensor_stream
    s = socketserver.TCPServer((host, port), SensorDataHandler)
  File "/home/siraj/miniconda3/envs/auto-rccar/lib/python3.7/socketserver.py", line 452, in __init__
    self.server_bind()
  File "/home/siraj/miniconda3/envs/auto-rccar/lib/python3.7/socketserver.py", line 466, in server_bind
    self.socket.bind(self.server_address)
OSError: [Errno 99] Cannot assign requested address

    self.socket.bind(self.server_address)
OSError: [Errno 99] Cannot assign requested address

hamuchiwa, 2019-03-25:
@Sirajabudabe Address is taken by your previous run, make sure to exit the program, or you can close the terminal/pycharm, then the address will be available.

closed 2019-03-25 by hamuchiwa

Prologueyan, 2019-05-18:
how did u solved this problem？