hamuchiwa/AutoRCCar #178: ERROR OF BROKEN PIPE WHILE TESTING ULTRASONIC SENSOR_CLIENT.PY

Devanshice020, 2019-03-26:
[image]
[image]
 i am facing this error of broken pipe while testing the client and server side of ultrasonic sensor.. it counts upto around 20 measurements and perfectly streams to the server side and then there is this error in the client side . please can you look into it @hamuchiwa , it is very urgent .

hamuchiwa, 2019-03-26:
@Devanshice020 Because server only runs for 10 second, so you see total 20 values printed (1/0.5*10=20). When server exits, you get a broken pipe error on the client side.

SaiSurya9999, 2019-03-27:
@hamuchiwa please respond to #177.

Devanshice020, 2019-03-27:
so @hamuchiwa  how do i resolve my problem sir?..i want to get measurements as long as i want as in the final testing , the car has to detect the obstacle  till i manually shut it off? how do i extend the server time to infinite or some thing? pls help sir @hamuchiwa

hamuchiwa, 2019-03-27:
@Devanshice020 Just remove line 32, 32 from `ultrasonic_server_test.py`.

closed 2019-03-27 by hamuchiwa