hamuchiwa/AutoRCCar #182: Issue with rc_driver_nn_only.py : 'NoneType' object has no attribute 'argmax'

VishalWrench, 2019-03-27:
@hamuchiwa I am doing this project without raspberry pi and using android phone(IP Camera app)to stream the track. I successfully captured the images and trained  my model but when i load my model and run rc_driver_nn_only.py it gives me this error.I have provided the image of the error that i got.help me with that
[image]

VishalWrench, 2019-03-27:
### this is my rc_driver_nn_only.py  
import cv2
import numpy as np
import socket
import serial
from model import NeuralNetwork
from rc_driver_helper import RCControl
import urllib
url = 'http://100.65.158.130:8080/video'  
stream = urllib.urlopen(url)
stream_bytes = ''

class RCDriverNNOnly(object):

    def __init__(self,serial_port):

        #self.server_socket = socket.socket()
        #self.server_socket.bind((host, port))
        #self.server_socket.listen(0)

        # accept a single connection
        #self.connection = self.server_socket.accept()[0].makefile('rb')

        # load trained neural network
        self.nn = NeuralNetwork()
        self.nn.load_model("C:/Users/Gireesh/AutoRCCar-master-Zheng/computer/saved_model/f_model.xml")
        self.send_inst = True
        #self.ser = serial.Serial("COM3", 115200, timeout=1)
        self.rc_car = RCControl(serial_port)

    def drive(self):
        stream_bytes = ' '
        image_array = np.empty((1,384000))
        try:
            # stream video frames one by one
            while self.send_inst:
                stream_bytes += stream.read(1024)
                first = stream_bytes.find(b'\xff\xd8')
                last = stream_bytes.find(b'\xff\xd9')

                if first != -1 and last != -1:
                    jpg = stream_bytes[first:last + 2]
                    stream_bytes = stream_bytes[last + 2:]
                    gray = cv2.imdecode(np.frombuffer(jpg, dtype=np.uint8), cv2.IMREAD_GRAYSCALE)
                    image = cv2.imdecode(np.frombuffer(jpg, dtype=np.uint8), cv2.IMREAD_COLOR)

                    # lower half of the image
                    #height, width = gray.shape
                    #roi = gray[int(height/2):height, :]

                    cv2.imshow('image', image)
                    # cv2.imshow('mlp_image', roi)

                    # reshape image
                    #image_array = roi.reshape(1, int(height/2) * width).astype(np.float32)
                    image_array = np.ravel(np.array(gray).astype(np.float32))
                    print(image_array)
                    print(image_array.dtype)
                    #print(image_array)
                    # neural network makes prediction
                    prediction = self.nn.predict(image_array)
                    print(prediction)
                    self.rc_car.steer(prediction)

                    if cv2.waitKey(1) & 0xFF == ord('q'):
                        print("car stopped")
                        self.rc_car.stop()
                        break
        finally:
            cv2.destroyAllWindows()

            #self.connection.close()
            #self.server_socket.close()

if __name__ == '__main__':
    # host, port
    #h, p = "192.168.1.100", 8000

    # serial port
    sp = "COM3"

    # model path
    #path = "C:/Users/Gireesh/AutoRCCar-master-Zheng/computer/saved_model/night_model.xml"
    #RCDriverNNOnly()
    rc = RCDriverNNOnly(sp)
    rc.drive()

hamuchiwa, 2019-03-27:
@VishalWrench It seems either `image_array` is empty or model is not loaded. Can you check if `self.nn.model` is None?

VishalWrench, 2019-03-28:
@hamuchiwa i have printed image_array it is not empty and i have checked self.nn.model it is also not returning None
i have added if statement as below to check whether the model is loaded or not
if self.nn.model == None:
    print('fail')
    break
i has not printed fail so model is loaded

VishalWrench, 2019-03-29:
@hamuchiwa now i have solved that problem but when i have trained it have given 
train accuracy:86%
validation accuracy:79%
when i have trained it predicted all directions
but when i run my rc_control_nn_only.py it is  only predicting forward i.e 0 help me with this
[image]

hamuchiwa, 2019-03-29:
@VishalWrench How did you solve the issue? Regarding the training, you can take a look #168, #173.

VishalWrench, 2019-03-29:
i have trained the model as a list so i have change the input image as
image = np.array( [ np.ravel(image_array) ] )

VishalWrench, 2019-03-29:
### this is my code
import cv2
import numpy as np
import socket
import serial
from model import NeuralNetwork
from rc_driver_helper import RCControl
import urllib
url = 'http://100.96.36.26:8080/video'  # duzenlenecek
stream = urllib.urlopen(url)
stream_bytes = ''
class RCDriverNNOnly(object):

    def __init__(self,serial_port):

        #self.server_socket = socket.socket()
        #self.server_socket.bind((host, port))
        #self.server_socket.listen(0)

        # accept a single connection
        #self.connection = self.server_socket.accept()[0].makefile('rb')

        # load trained neural network
        self.nn = NeuralNetwork()
        self.nn.load_model("C:/Users/Gireesh/AutoRCCar-master-Zheng/computer/saved_model/29_model.xml")
        self.send_inst = True
        
        self.rc_car = RCControl(serial_port)

    def drive(self):
        stream_bytes = ' '

        try:
            # stream video frames one by one
            while self.send_inst:
                stream_bytes += stream.read(1024)
                first = stream_bytes.find(b'\xff\xd8')
                last = stream_bytes.find(b'\xff\xd9')

                if first != -1 and last != -1:
                    jpg = stream_bytes[first:last + 2]
                    stream_bytes = stream_bytes[last + 2:]
                    gray = cv2.imdecode(np.frombuffer(jpg, dtype=np.uint8), cv2.IMREAD_GRAYSCALE)
                   
                  

                    cv2.imshow('image', gray)
                    

                   
                    image_array = gray.astype(np.float32)
                  
                    image = np.array([np.ravel(image_array)])
                    print(image)
                  
                  
                    prediction = self.nn.predict(image)
                    print(prediction)

                    self.rc_car.steer(prediction)

                    if cv2.waitKey(1) & 0xFF == ord('q'):
                        print("car stopped")
                        self.rc_car.stop()
                        break
        finally:
            cv2.destroyAllWindows()

            #self.connection.close()
            #self.server_socket.close()

if __name__ == '__main__':
    # host, port
    #h, p = "192.168.1.100", 8000

    # serial port
    sp = "COM3"

    # model path
    #path = "C:/Users/Gireesh/AutoRCCar-master-Zheng/computer/saved_model/night_model.xml"
    #RCDriverNNOnly()
    rc = RCDriverNNOnly(sp)
    rc.drive()

closed 2019-04-12 by hamuchiwa