hamuchiwa/AutoRCCar #183: CONTROLS ARE NOT WORKING PROPERLY.

Devanshice020, 2019-03-29:
Hello Mr. Zheng Wang, i am facing an issue regarding the controls of the car when run in the collect_data.py program. Controls given by me and what the car is behaving is as follows:
1)forward left == reverse right
2)forward right == reverse left
3) reverse left == forward
4) reverse right == reverse
PLEASE SEE: when we are controlling through arduino in our initial testing phase, the car is behaving properly i.e command given by us = what car reacts but these problem while attaching with RPI and collecting data and running , the car does not responds in a specific manner.
This problem has become a constant whenever we restart or work on out project the next day. Please can you suggest what can we do? 
I have thorougly checked all the connections and they are correct as the arduino testing is correctly behaved .
please help sir @hamuchiwa .
thanks in advance.

hamuchiwa, 2019-03-29:
@Devanshice020 Does `rc_control_test.py` work for you?

Devanshice020, 2019-03-29:
yes it ran peefectly. but i facing this error while running this connecting to rpi and collecting data.. and today was almost 10th time that i am having this problem. what could possibly the reason for it?should i change anything in the code? @hamuchiwa . can you suggest sir

Devanshice020, 2019-03-31:
Issue has solved.
I have changed pin high from low and low to high in rc_keypad_control.ino
Thanks @hamuchiwa for your response and help.

closed 2019-03-31 by Devanshice020