hamuchiwa/AutoRCCar #184: Problem in movement of car

ShwetaLalwani, 2019-03-30:
Hi,
While running collect_training_data.py,everything is working fine. When I'm placing my car in hands, wheels are properly moving in forward reverse left right direction,... But when I'm placing my car on floor , it is only moving in forward and reverse direction. What must be the problem?

VishalWrench, 2019-03-30:
It might have low battery( change batteries) or it can't bear the weight u
put on them like pi and power bank.

ShwetaLalwani, 2019-04-06:
Hi, after running rc_driven_nn_only.py ,I'm getting this error- module object has no attribute ml.
What should I do to remove this ?

VishalWrench, 2019-04-06:
Attach the screenshot of whole error u got

ShwetaLalwani, 2019-04-06:
[image]

VishalWrench, 2019-04-06:
It may be open cv version issue problem try to downgrade opencv

ShwetaLalwani, 2019-04-06:
Downgrade to which version? Now my OpenCV version is 3.4.4

ShwetaLalwani, 2019-04-09:
Ok that error is solved. Now I'm getting this error-.[
[image]
](url)

ShwetaLalwani, 2019-04-20:
My car is moving only in forward direction. Plz help. @hamuchiwa

sundar19, 2019-05-29:
@ShwetaLalwani rc_driver_nn_only.py  should be run on your computer and not RPi

arpitpartani, 2019-06-26:
while running collect_training_data.py my car is not moving with computer arrow buttons , hence not collecting frames

arpitpartani, 2019-06-26:
pls help @ShwetaLalwani @VishalWrench

ShwetaLalwani, 2019-06-26:
@arpitpartani first you have to run collect_training_data.py and then in other terminal run stream_client.py ,then a pygame window along with a streaming window will pop up. Click on pygame window and then click the arrow keys to move your car. When your data collection is done press x or q to exit from pygame window and then press q to exit from streaming window , your saved frames and lost frames will then be displayed on terminal window.

arpitpartani, 2019-06-26:
exactly we are doing @ShwetaLalwani 
but its not working

can u share your collect_training_data.py code

ShwetaLalwani, 2019-06-26:
@arpitpartani actually I've done my whole project on raspberry pi, I haven't used Arduino. So my code is a bit different than this, but the procedure is same . I've only changed the pins according to my dc motor pins and removed the Arduino part. Rest is same as collect_training_data.py. Tell me what error are u getting?

arpitpartani, 2019-06-26:
Atleast send the code

Yoganandhan01, 2020-03-20:
@arpitpartani I am also having the same problem after running collect_training_data.py file and model_training.py after I'm running the rc_driver.py my car only goes on straight line it does not follow the lane..?

Yoganandhan01, 2020-03-20:
@arpitpartani can you explain the file "train_predict_test.ipynb" the jupyter notebook file in model_train_test folder..?