hamuchiwa/AutoRCCar #185: error while running rc_drive.py

Devanshice020, 2019-03-31:
An exception has occurred, use %tb to see the full traceback.

SystemExit

C:\Users\Krishna\Anaconda3\lib\site-packages\IPython\core\interactiveshell.py:3304: UserWarning: To exit: use 'exit', 'quit', or Ctrl-D.
  warn("To exit: use 'exit', 'quit', or Ctrl-D.", stacklevel=1)

I am getting this error while running rc_driver.py

SaiSurya9999, 2019-04-01:
It may be libraries issue do run environment.yml file to create environment. And then activate it before execution. Even though if problem exist compare the versions with repository versions.

hamuchiwa, 2019-04-01:
@Devanshice020 Does it say what exception?

Devanshice020, 2019-04-01:
File "C:/Users/Krishna/Desktop/RPI/AutoRCCar-master/computer/rc_driver.py", line 160, in handle
    sys.exit()

SystemExit

It shows this error while i run %tb in shell

hamuchiwa, 2019-04-01:
@Devanshice020 Run the code in a terminal instead, sys.eixt() will exit python.

Devanshice020, 2019-04-01:
Thanks @hamuchiwa for your help and support I have solved this issue by importing every function from rc_driver_help.py manually.
And also by defining 
d_sensor_thresh = 30
d_stop_light_thresh = 25
d_stop_sign = d_stop_light_thresh
d_light = d_stop_light_thresh
outside every function.

apoorvyadav1111, 2019-04-20:
@hamuchiwa i have same issue. It says exception occurred and system exit. Please help

hamuchiwa, 2019-04-22:
@apoorvyadav1111 @Devanshice020 #174

closed 2019-05-18 by hamuchiwa