hamuchiwa/AutoRCCar #188: Serial output too fast

MyFirstRoboticProject, 2019-04-03:
@hamachiwa
When running the rc_driver.py (or nn only), the forward output responses on a straight track comes too fast and quickly (I get 10 forward responses in just 1 second). Due to this my car is unable to keep up and even if I terminate the code (i.e. rc_driver.py) it keeps on running straight unless I plug out the arduino cable. Any solution?

VishalWrench, 2019-04-03:
yeah buddy i'm facing the same issue the frames are slow and the prediction and frames are out of sync . when i slowly move my camera around the track it predicts well. Solution is either to increase the frame rate or to reduce the car speed

MyFirstRoboticProject, 2019-04-03:
Slowing the car will make the problem worse. My car is slow enough, but the serial output rate is just too fast. I need a way to reduce the rate of serial output (like three forwards per second or three forward rights per second). Also my camera is fine and the frame rate is satisfactory. The only problem is the serial output rate.

MyFirstRoboticProject, 2019-04-03:
@VishalWrench BTW did you successfully train your car?

MyFirstRoboticProject, 2019-04-04:
I found the solution to my problem. Closing thread

closed 2019-04-04 by MyFirstRoboticProject

VishalWrench, 2019-04-04:
How did u solved it??

MyFirstRoboticProject, 2019-04-04:
Well, the output serial data would change instantaneously regardless how fast it was, so it didn't matter. And also if you want to achieve a greater accuracy for training make the driving path as narrow as possible.

Devanshice020, 2019-04-05:
have you made your car by your self? @MyFirstRoboticProject

MyFirstRoboticProject, 2019-04-05:
No. Used a normal rc car

Devanshice020, 2019-04-05:
Have you code for cr's motor to control it's speed? @MyFirsrRoboticProject

MyFirstRoboticProject, 2019-04-05:
Yup. Not an ideal solution though. I'm trying to find some other way to slow my car's speed (other than using a potentiometer or a resistor)

Devanshice020, 2019-04-05:
I am having the same problem everything is working properly but car is very fast so it's not working properly.
If you find the solution plz let me know.
Thanks for your help

MyFirstRoboticProject, 2019-04-05:
Ok